g0orx/pihpsdr #2: CW

ve3mm, 2016-07-18:
Hi John

I built your latest code and find it super to be able to use my RPi 3 with Hermes.

I wanted to let you know that there are a couple of things on cw that make it difficult to use:
- There is no offset on cw when transmitting.  ie the cw is being sent at 0 Hz offset rather than at the sidetone offset.
- There is a short buzz on the receive audio when switching back to receive from transmit on cw.
- The weighting on cw between the dashes and dots is too short.  I did not see a control for this setting.  Can you hard code the value to 50 if you are not going to put in a setting adjustment for it.

If you point me in the right direction I will try and make the changes to the code myself.

73

Rick ve3mm

radi8, 2016-07-19:
A further improvement would be to have the carrier centre in the filter passband in CW mode when it is clicked on in the spectrum or spectrogram. At present clicking on a cw signal puts it at zero beat and it is necessary to roll the mouse wheel to bring it into the passband. This is turning out to be a wonderful radio and I am very keen to see it support Hermes-Lite and particularly to get some audio.

73, Graeme zl2apv

g0orx, 2016-07-19:
Hi Rick,

Please remember this is still very much a Beta version of the code and 
there are still a lot of changes being made.   Our main goal was to get 
a small Transceiver working using Hermes/Apollo.

CW has been low on the list but there are a number of changes that we 
are about to push so if you can wait for another few days then I think 
you will see all the suggestions you have made.

Our recent push has been for both the integration of FreeDV and the new 
Ethernet protocol.

-- John

radi8, 2016-07-19:
Hello John,

That was quick and thank you very much. Not just for the CW enhancements but also for all the work you have done over time as I have been following your projects for a number of years. I did my first C programming on the ghpsdr3-alex fork and by following your code style and ideas I learned more than a hundred books could have taught me. It is a great pity the fork diverged too far to be able to be merged with your original work. I still did not get exactly what I was after which was minimal latency on a local lan with a server doing the crunching like dspserver and minimal lan traffic to my remote computer.

For a beta this radio is going pretty good. The finished job is going to be fantastic.

73 es tu, Graeme zl2apv

g0orx, 2016-07-19:
Hi Graeme,

most of the changes were already done but not pushed yet so I decided to 
just finish them off and push them.  Not a lot of testing so beware!

I am working on a version of pihpsdr with a server running at the HPSDR 
device on an SBC and a remote client which will be an option in pihpsdr 
to run in a remote mode.

I did get the OK from Flex to build a server that would allow the 
Maestro to connect to HPSDR so have been looking at that as a protocol, 
but no final decision yet.  Too many other things going on ...

-- John

ve3mm, 2016-07-20:
Hello John

Thank you for making the changes to the code in response to my request.  I downloaded and rebuilt the files. 

The cw weighting is great now with the adjustable settings.  I have set it to 50 and it sounds great.

I am not getting the proper offset on transmit, though.  It is still transmitting on zero offset.  I see the build date is July 19 so I think I have all the right files.  

I also have the short buzz sound on return to receive from transmit.

Maybe Graeme can chime in and let me know if his offset issue is fixed.

I totally understand that this program is under development and understand that you have your own priorities on what the timing changes should have.   

I appreciate and thank you for all of the work that you have shared with the hpsdr community over the years. 

The other versions of pihpsdr that you are working on sound exciting too.

73

Rick ve3mm

radi8, 2016-07-20:
Hello Rick and John,

No the offset issue is not fixed yet. It still places the signal at the carrier frequency instead of offsetting the receiver by the cw tone. I have placed below the code I used in ghpsdr3-alex to put the signal into the passband. In CW mode the receiver is not displaying the carrier frequency but the frequency of the centre of the passband and the carrier is offset by the tone frequency + or -

void UI::frequencyChanged(long long f) {
    QString command;
    long long freqOffset = f; //Normally no offset (only for CW Rx mode)

    frequency=f;
    if((mode.getStringMode()=="CWU")&&(!widget.vfoFrame->getPtt())){
        freqOffset-=cwPitch;
    }
    if((mode.getStringMode()=="CWL")&&(!widget.vfoFrame->getPtt())){
        freqOffset+=cwPitch;
    }
    //Send command to server
    command.clear();
    QTextStream(&command) << "setFrequency " << freqOffset;
    connection.sendCommand(command);
    //Adjust all frequency displays & Check for exiting current band
    band.setFrequency(frequency);
    widget.spectrumView->setFrequency(frequency);
    widget.vfoFrame->setFrequency(frequency);
//    widget.waterfallView->setFrequency(frequency);
}
Hope the code shows better what I mean than my explaination John.

I have found a couple of other things which you probably aware of and have a note to fix them in the future but if you would like me to record them as issues I would be happy to do so.

Again thanks for everything,
Graeme ZL2APV

radi8, 2016-07-20:
I have just realised that the issue is partially fixed. When clicking on the spectrum it works fine but when clicking on the waterfall it still puts the signal on the carrier frequency. Please disregard the previous post as I can see you are already on to it and the waterfall will happen in due course. Sorry I did not test it fully.

73, Graeme

g0orx, 2016-07-20:
I have pushed a new version with the waterfall fixed when clicked in CW 
mode - sorry, forgot to update it.

Just out of interest we are not usually using a mouse, just a 7 inch 
touch screen and encoders for the VFO, gains, etc and push buttons for 
the other functions.

We have a list of bugs/enhancements that we are working on that should 
be completed by the end of this month.  A large amount of time has 
recently been getting the PCB layout and prototypes built and tested. 
Also, I am working on two other projects for HPSDR including enhancing 
the code for the Apollo amplifier/tuner.

FYI, if you are running on a RPi then there is no reason to build the 
source every time.  There is a tar file in the release directory with 
all the binaries, etc, needed to run the application.

use the following to get the code:

wget https://github.com/g0orx/pihpsdr/raw/master/release/pihpsdr.tar

and the install instrictions:

wget 
https://github.com/g0orx/pihpsdr/raw/master/release/documentation/pihpsdr-install.pdf

-- John

radi8, 2016-07-20:
Thanks for the waterfall update John, it is now working perfectly. Yes I understand that my use of the radio is not how it was intended as I am using Linux Mint 17.3 on a pretty standard desktop machine but even so the radio works really well with a mouse and in this configuration I really would not wish for much different.

Being a cw operator mostly, one thing that would be nice is for the display to be fixed and the cursor follows where you click or drag. While running a session on air you get used to where the stations are operating and if you slide the screen this is immediately lost. Two arrow buttons or a click at the screen extremities would jump to the next frequency segment. Ideally you could switch between the 2 modes. It sort of happened on ghpsdr3 by using Rx2 but don't be sidetracked by my wishlist. This radio in its simplicity is one of the best so far.

73, Graeme

ve3mm, 2016-07-20:
Hi John

Thanks again for the update.  I did download the tar binary this time. It is a lot quicker to install it.

Results:
- I confirm clicking with a mouse or my finger on the display now moves the cw signal into the passband at the correct cw offset.  

- CW transmit is still off frequency. 
ie.  CWU, sidetone 500 hz., station tuned in correctly.  When you transmit you are transmitting 500 hz low and the other station normally will not hear you as you are out of their passband. 
The same issue only reversed  (transmit 500 hz. high) occurs on CWL.  

Sorry to be a pest on this.

73

Rick ve3mm

g0orx, 2016-07-22:
I have just pushed an update that "should" fix the CW transmit offset. 
Actually it was the receive frequency that needed to be offset when in 
CW mode.  I had a problem that took a while to find because I was not 
setting the duplex bit in the protocol so it was not using the separate 
tx and rx frequencies that were being sent in the protocol.

I still need to work on the "short buzz" sound when going from TX to 
RX.  It was longer some time ago and it got improved but not completely 
removed.  I need to look at the WDSP configuration parameters again to 
get it adjusted correctly.

-- John

ve3mm, 2016-07-22:
Hi John

I am sorry to report that the offset problem is not fixed.  The frequency offset between rx and tx is still there on CWU and CWL.

Thanks for sticking at it, but I wouldn't blame you if you went on to other things.

73
Rick ve3mm

closed 2016-07-22 by ve3mm

g0orx, 2016-07-22:
Interesting as I  am seeing it working ok.  I will check again tomorrow but
all the testing I did today showed it was working as PowerSDR works. I
tested it with both a Hermes and an Angelia.

Can you let me gave the settings you are using?  Sample rate, filter, cw
sometime frequency?

I do want this to be fixed so don't worry about it being a problem.

-- John