g0orx/pihpsdr #3: CW offsets

radi8, 2016-07-22:
Hi John,

Sorry to say there are still problems with the CW offset. All tests are made in CW mode.
1. Rx is working perfectly i.e. setting the vfo to say 14.02 and transmitting from another radio on 14.02 the signal is set exactly in the centre of the passband.

2. On pihpsdr setting the Tune to max (255) and vfo to 14.02 the tx tone is at 14.02 plus the 700 Hz tone offset, 14.0207. Of course it should have been at 14.02

3. I started to reduce the Tune level while looking at the spectrum from my Hermes-Lite it immediately produced a string of spurious products which went back to a single tone when I raised the output level back to 255.

4. After a time the carrier shifted to +400 Hz and the spurs disappeared when adjusting the tune level but the control became less effective and eventually the signal level dropped by about 40 dB.

5. I figured that the radio had got into a tangle and restarted pihpsdr whereupon everything returned to condition 1 so something is not stable plus the cw Tx has the same offset as the Rx.

I compiled using N=x86 in the Makefile but will try using N=up and report if there is any change. Another thing I need to check is that as I was using Quisk with HiQSDR for monitoring pihpsdr and it was running on the same computer as pihpsdr and there could have been some interaction so I will repeat the tests with different computers.

I just tried going to AM and keying the Tx. First key I got the spectrum with the line of spurs and going back to Rx gave me the buzz we get on CW. After that AM behaved normally each time I triggered it to Tx. I was using MOX rather than tune  for this test. Tried going to CW and back to am but fault did not repeat.

Hermes-Lite does not have the CW lines set up yet so I have been using tune rather than MOX to get my transmit tone. Of course if you have an offset on tune then my findings are invalid apart from the spectrum spurs. Hope this is of some help.

73, Graeme ZL2APV

radi8, 2016-07-22:
I need to add that I gave it one last try and it transmitted the correct frequency tone but after 2 or 3 seconds it jumped up 400 Hz and dropped a bit in level. It was all so quick I did not get all the data but something is very intermittent. The spurs are there with the tune slider down below 22.

73

g0orx, 2016-07-23:
Hi Graeme,

Interesting. I have received a couple of emails from Ken, N9VV, saying 
he has now been able to make successfully several CW QSO's using the new 
code. He is using an ANAN-100D.  I have an ANAN-100D, Hermes/Apollo and 
a Orion MK 2 board to test with.

In tune mode the CW signal is not being created by the firmware, but by 
the code in the RPi and that is transmitted on the center of the filter 
passband so that would explain why  you are seeing that offset.  I will 
look at changing that when tuning in CW mode.

One thing to check is the power supply to the RPi.  We have seen 
unpredictable results when the supply is not up to the current that is 
being drawn. Do you ever see the low voltage block appear in the top 
right of the display.  If this appears then the RPi will disable turbo 
mode resulting in a drop in performance that may well effect the

-- John

g0orx, 2016-07-23:
> I will have a look at the Tune code as there may be a problem there rather
> than in normal CW mode.

Forgot to say that compiling with the different UNAME_N options is just to
change the platform it is being compiled on for the GPIO use. The up option
is to tell the GPIO code that it is running on an UP-Board. The x86 option
basically disables GPIO.

-- John

radi8, 2016-07-23:
Hello John,

I hope I am not adding a level of confusion here as I am not using an Rpi but a 64 bit desktop dual core running Linux Mint 17.3 so perhaps the bugs I see are caused by running outside the current scope of the project. Thanks for the explaination of the UNAME_N options so I will stay with x86 until I get a Rpi setup.

I am happy to continue testing on my desktop but if my reports are confusing the issue I am also happy to follow the project silently until you get to desktop testing stage. Again a very big thank you for giving so much of your time and talent to the SDR community,

73, Graeme zl2apv

g0orx, 2016-07-23:
Hi Graeme,

No problem.  I will try to put some time into more testing on a desktop. I
plan on updating ghpsdr for desktop systems to use the latest wdsp and the
new protocol, so maybe that would be the one to wait for. It will use all
the network code from the RPi code but will support multiple receivers and
have the more complex UI.

FYi Ken, N9VV, is running on a desktop with no problems. I have just added
RIT that he has been requesting especially for CW.

I will also look at tune in CW mode as i do need to resolve that.

-- John

ve3mm, 2016-07-23:
Hi John

I just downloaded version 2016-07-22 5f7bfbf55-dirty (tar) and am happy to report that I made the first cw qso on 20m with UR3HC!

The cw offset issue is solved now.  Thank you.

The popping (single tone now, used to be a buzz) is still there on the transition from tx to rx.  I have to use a turnaround time of 260 ms or longer to not be bothered by it. 

Set up is:
RPi 3 with 7 inch touchscreen
Hermes with Alex filter for rx only
Hardrock50 amp for tx

I am curious as to why you are not using the Hermes FPGA for cw?

73 Rick ve3mm

g0orx, 2016-07-23:
Good to hear you gave made a successful QSO.

Why do you think I  not using the FPGA CW?  I am!

-- John

ve3mm, 2016-07-23:
Hi John

I just finished up a run of about a dozen EU cw qsos on 20m  Really solid now.  Audio is great.  Lots of fun with such a little computer.  Thanks for putting it together!

I went back and re-read your earlier post to Graeme and realized that you only said that the tune function was not using the firmware.  I understand now. 

73
Rick

g0orx, 2016-07-23:
Hi Rick,

That is good to hear. Glad you are enjoying it.

-- John

ve3mm, 2016-07-29:
Hi John

Just an update. I downloaded your latest commit.  

There is a definite improvement in the transition from tx to rx on cw.  The buzz / tone is not audible any longer at delays down to 50 ms.  At this setting the relay in Alex is clacking away.  Thanks for this improvement.

I get around this by combining Hermes with a Hardrock50 amp which has pin diode qsk rx to tx switching (no Alex).  The filters in the Hardrock 50 aren't as good as the ones in Alex, but the qsk is much faster allowing transition times as fast as the settings allow.

Do you have any plans to implement the low latency filters at some point, as this would allow fast qsk and lower overall latency for the ssb and digitial guys?

73

Rick ve3mm

g0orx, 2016-07-29:
Hi Rick,

Pleased to know you can see some improvement.  There are some more changes
coming soon. I have changed the tune mode to use the WDSP functions rather
than my own code.

I do plan on adding the low latency filter code. I have just not had time
to port the latest version of wdsp to Linux yet.

I am also working on implementing psk digital modes into the application. I
have receive working with text currently output to stdout. I have some work
to do on the UI for this mode.

-- John

closed 2016-11-23 by radi8