g0orx/pihpsdr #4: Compile error with psk

radi8, 2016-08-06:
Hello John,

Thanks for the next step in the radio. I am not sure if you are aware of the following error when compiling with PSK_INCLUDE=PSK uncommented. I could not find a pskcoresdr.h anywhere.

gcc -g -D x86   -D FREEDV -D PSK -D GIT_DATE='"2016-08-05"' -D GIT_VERSION='"v1.0.0-beta-2-g5d54683-dirty"' -O3 `pkg-config --cflags gtk+-3.0` -c -o freedv.o freedv.c
gcc -g -D x86   -D FREEDV -D PSK -D GIT_DATE='"2016-08-05"' -D GIT_VERSION='"v1.0.0-beta-2-g5d54683-dirty"' -O3 `pkg-config --cflags gtk+-3.0` -c -o gpio.o gpio.c
gcc -g -D x86   -D FREEDV -D PSK -D GIT_DATE='"2016-08-05"' -D GIT_VERSION='"v1.0.0-beta-2-g5d54683-dirty"' -O3 `pkg-config --cflags gtk+-3.0` -c -o psk.o psk.c
psk.c:5:24: fatal error: pskcoresdr.h: No such file or directory
 #include <pskcoresdr.h>
                        ^
compilation terminated.
make: *** [psk.o] Error 1
gvj@gvj-Mint ~/Programs/pihpsdr $ 

73, Graeme ZL2APV

g0orx, 2016-08-06:
Please read the latest pihpsdr-build.pdf document which shows how to get
the psk source code and compile the shared library.

-- John

radi8, 2016-08-06:
I'm really sorry about that John, I read the .odt file and when I got to the whitespace at the end of the FreeDV section I never thought to page down one more.

There is a runtime issue for me similar to what I had for FreeDV where it could not find the library so I put a copy from the codec2 default directory (x86_64-linux-gnu) into /usr/local/lib. In this case libpsk.so installs there by default but still can't be found even running as root.
gvj@gvj-Mint ~/Programs/pihpsdr $ sudo ./pihpsdr
[sudo] password for gvj: 
./pihpsdr: error while loading shared libraries: libpsk.so: cannot open shared object file: No such file or directory
gvj@gvj-Mint ~/Programs/pihpsdr $ ll /usr/local/lib/
total 3992
drwxr-xr-x  5 root root     4096 Aug  6 21:31 ./
drwxr-xr-x 11 root root     4096 Jun 11  2010 ../
-rw-r--r--  1 root root   577110 Oct  6  2015 libcodec2.so
-rw-r--r--  1 root root   577110 Oct  6  2015 libcodec2.so.0.5
-rwxr-xr-x  1 root root    89826 Aug  6 21:31 libpsk.so*
-rwxr-xr-x  1 root root  2821731 Jul 16 15:17 libwdsp.so*
drwxrwsr-x  4 root staff    4096 Sep 14  2015 python2.7/
drwxrwsr-x  3 root staff    4096 Jun 28  2015 python3.4/
drwxr-xr-x  2 root root     4096 Oct  1  2015 x86_64-linux-gnu/
gvj@gvj-Mint ~/Programs/pihpsdr $ 

I tried on a 32 bit system with FreeDV and even placing the libraries in /usr/local/lib was not enough and I had to manually add the directory to the library path each time I ran a session. I have not yet tried psk on this machine.

Many thanks, Graeme

g0orx, 2016-08-06:
did you try:

sudo ldconfig

-- John

radi8, 2016-08-06:
I completely rebooted the machine after runtime error but still the same.

.. Graeme

radi8, 2016-08-10:
A second reboot and it all worked.

closed 2016-08-10 by radi8