g0orx/pihpsdr #5: Please consider renaming and generalizing the LimeSDR support for all SoapySDR devices.

Analias, 2016-08-06:
I saw your project and was intrigued by the SoapySDR support for the LimeSDR. Using your code, I was able to get pihpsdr to compile and run under Mac OS X (El Capitan) and use it with my SDRPlay. It's still rough, and I'm running into an error with the miri SoapySDR driver setting a hardware register incorrectly, but I expect to be able to solve it.

Most of your code in lime_protocol.c and lime_discovery.c is pretty generic SoapySDR code. In fact, early on I had pointed to the wrong SoapySDR device and discovered that pihpsdr had opened and was using the SoapySDR support for the built-in mic on my Mac.

Please consider renaming and generalizing the LimeSDR support for all SoapySDR devices.

-Freeman

KC7NOA, 2017-02-18:
Have you tried the new SoapySDR code with PiHPSDR ?

g0orx, 2017-02-18:
Sorry, have not had any spare time to work on SoapySDR.  Too much to do with the HPSDR devices and development work on the 8000DLE.
This is open source so anyone can help make changes if you are able by making a branch and then requesting a merge.

closed 2017-02-18 by g0orx

KC7NOA, 2017-02-18:
I dont think the problem had anything to do with pihpsdr .. soapysdr LimeSdr was not working properly until just recently ...