g0orx/pihpsdr #6: Compile error

radi8, 2016-08-24:
Hello John,

When compiling the following error pops up

gvj@gvj-Mint ~/Programs/pihpsdr $ make
rm -f version.o
gcc -g -D x86   -D FREEDV -D PSK -D GIT_DATE='"2016-08-23"' -D GIT_VERSION='"v1.0.0-beta-18-g84a142c-dirty"'  -O3 `pkg-config --cflags gtk+-3.0` -c -o audio.o audio.c
In file included from audio.c:35:0:
radio.h:42:1: error: unknown type name ‘DISCOVERED’
 extern DISCOVERED *radio;
 ^
make: *** [audio.o] Error 1
gvj@gvj-Mint ~/Programs/pihpsdr $

Looking for instances of DISCOVERED

gvj@gvj-Mint ~/Programs/pihpsdr $ grep -r DISCOVERED
main.c:        DISCOVERED *d=&discovered[selected_device];
main.c:  DISCOVERED* d;
discovered.h:struct _DISCOVERED {
discovered.h:typedef struct _DISCOVERED DISCOVERED;
discovered.h:extern DISCOVERED discovered[MAX_DEVICES];
new_protocol_programmer.c:    DISCOVERED* d=&discovered[selected_device];
discovered.c:DISCOVERED discovered[MAX_DEVICES];
radio.c:DISCOVERED *radio;
radio.h:extern DISCOVERED *radio;
Binary file release/pihpsdr/pihpsdr matches
Binary file release/pihpsdr.tar matches
gvj@gvj-Mint ~/Programs/pihpsdr $

The structure seems to be defined OK and I can't see any problems with the code but then I am no programmer. Hope you can help.

73, Graeme zl2apv

g0orx, 2016-08-25:
Sorry, the latest push (beta-22) fixes the problem.

-- John

radi8, 2016-08-25:
Thanks John, it works beautifully.

closed 2016-08-25 by radi8