g0orx/pihpsdr #7: Waterfall shift not matching graph frequency shift on mouse click

radi8, 2016-08-27:
Hello John

When clicking on the spectrum display the frequency shifts correctly to the bandpass under all conditions but the waterfall only shifts correctly at 48000 sample rate. At all other sample rates the step is too small depending on sample rate selected.

BTW is Tx audio meant to work on the desktop compile? I get an output on tune but nothing from the microphone but maybe I am trying to get ahead of your coding.

73, Graeme zl2apv

g0orx, 2016-08-27:
Thanks,  I will check the waterfall.
There is no difference with the desktop an rpi for audio.  Are you connecting the mic to the radio or the computer? There is currently no support for microphone input from the computer.
-- John

radi8, 2016-08-27:
Thanks re the audio. Yes I am using a Hermes-Lite and need to connect audio via the computer so SSB will need to wait. No problem as I am 99% CW and I can do SSB on my XYL's TS-50 if I need it for now.

The radio is coming along extremely well and I am so impressed with the performance. WDSP seems to be noticeably improved on dttsp particularly with noise blanking and I seem to be able to run at 96000 sample rate without dropouts on my 802.11g wifi but I am still really looking forward to the day there is a dsp server and pihpsdr becomes a client so I can run full spec over wifi. Again thanks for the progress.

73, Graeme

radi8, 2016-09-02:
Thanks John, working just fine, great progress.

73, Graeme

closed 2016-09-02 by radi8