g0orx/pihpsdr #8: install error

yo2ldk, 2016-10-04:
Hi John,

  I have a Pi 3 board, and I try to install this nice and very useful software but, an error message make this impossible:
 cannot create regular file /root/.local/share/applications
I am log in as user pi ,make command sudo ./install.sh but the same result

please some tips to finish the install

 my best regards, 

         de yo2ldk.

g0orx, 2016-10-04:
Hi Alexander,
You should not be running 'sudo ./install.sh' , just './install.sh' as the pi user.
-- John

yo2ldk, 2016-10-04:
Hi Paul,

  Thank you for reply, yes, that I do it before, with the same result,not enough rights..
Anyway , all is ok now, I follow your steps from compile instructions and with that everything goes well.
 You can close this issue

 all the best an 73 !

closed 2016-10-04 by yo2ldk