g0orx/pihpsdr #9: Segfault on MOX or TUNE

radi8, 2016-11-06:
Hi John,

I get a segfault on MOX or TUNE which has happened since you changed to pulse. This is running on Linux Mint not a Raspberry Pi using a Hermes-Lite. Details below.

73, Graeme zl2apv

gvj@gvj-Mint ~/Programs/pihpsdr $ ./pihpsdr
Build: 2016-11-05 v1.0.0-beta-53-g3dbafa2-dirty
sysname: Linux
nodename: gvj-Mint
release: 3.19.0-32-generic
version: #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015
machine: x86_64
width=1920 height=1080
display_width=800 display_height=480
init
output_device: plughw:0,0 HDA NVidia
output_device: plughw:0,1 HDA NVidia
input_device: plughw:0,0 HDA NVidia
input_device: plughw:0,2 HDA NVidia
old_discovery
discover: looking for HPSDR devices on eth0
discover: bound to eth0
discover_receive_thread
discovered: received 60 bytes
discovery: found device=6 software_version=31 status=2 address=192.168.1.108 (00:1C:C0:A2:22:5D) on eth0
discovery: bytes read -1
discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
discovery: exiting discover_receive_thread
discover: exiting discover for eth0
discovery found 1 devices
discovery: found device=6 software_version=31 status=2 address=192.168.1.108 (00:1C:C0:A2:22:5D) on eth0
new_discover: looking for HPSDR devices on eth0
new_discover: bound to eth0 192.168.1.50 255.255.252.0
new_discover: bytes read -1
new_discover: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
new_discover: exiting new_discover_receive_thread
new_discover: exiting discover for eth0
new_discovery found 1 devices
discovery: found protocol=0 device=6 software_version=31 status=2 address=192.168.1.108 (00:1C:C0:A2:22:5D) on eth0
./pihpsdr: found 1 devices.
0x65f460 protocol=0 name=Hermes Lite
start_cb: 0x65f460
selected radio=0x65f460 device=6
loadProperties: 00-1C-C0-A2-22-5D.props
wdsp_init: rx=0 pixels=800 protocol=0
OpenChannel 0 buffer_size=1024 fft_size=4096 sample_rate=48000 dspRate=48000 outputRate=48000
OpenChannel 8 buffer_size=1024 fft_size=4096 sample_rate=48000 dspRate=48000 outputRate=48000
XCreateAnalyzer 0
SetAnalyzer channel=0 buffer_size=1024
SetAnalyzer channel=8 buffer_size=1024
SetRXAMode: rx=0 mode=0
SetTXAMode: tx=8 mode=0
SetAnalyzer channel=11 buffer_size=1024
old_protocol_init
audio_open_output: selected=0:plughw:0,0 HDA NVidia
audio_open_input: selected=0:plughw:0,0 HDA NVidia
old_protocol starting receive thread: buffer_size=1024 output_buffer_size=1024
old_protocol_init: prime radio
old_protocol: receive_thread
FPGA firmware version: 3.1
vfo_height=60
vfo_init: width=400 height=60
vfo_init: set Frequency,Mode,Filter
setMode: mode=0 m=1
setMode: 0 mode=0
SetRXAMode: rx=0 mode=1
setMode: 8 mode=0
SetTXAMode: tx=8 mode=1
setFilter: 150 2550
setFilter: filterLow=150 filterHigh=2550
setFilter: fl=150.000000 fh=2550.000000
menu_height=60
menu_init: width=200 height=60
input_device: plughw:0,0 HDA NVidia
input_device: plughw:0,2 HDA NVidia
output_device: plughw:0,0 HDA NVidia
output_device: plughw:0,1 HDA NVidia
meter_height=60
meter_init: width=200 height=60
panadapter_height=120
waterfall_height=180
sliders_height=90
sliders_init: width=800 height=90
toolbar_height=30
toolbar_init: width=800 height=30 button_width=100
vfo_configure_event_cb: width=400 height=60
meter_configure_event_cb: width=200 height=60
panadapter_configure_event_cb: width:800 height:120
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
full_tx_buffer: gain=32767.000000
Segmentation fault
gvj@gvj-Mint ~/Programs/pihpsdr $

g0orx, 2016-11-08:
> version: #3714.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015

Hi Graeme,

I cannot tell from the output where the segfault is.  However, there are 
a number of changes to the latest version of pihpsdr and wdsp on github 
that it would be worth testing with. This does work well with local 
audio in and out using pulse on the RPi.

If you are still getting the problem then try running pihpsdr with gdb 
and send me the output after running a bt command to get the stack trace.

-- John

radi8, 2016-11-09:
Hi John,

Thanks for your attention. Hope this output is OK as I have been sheltered behind Qt Creator's environment and never had to run gdb before. If you need more sing out. I will as a next step compile it on my RaspberryPi and see if it behaves there.

gvj@gvj-Mint ~/Programs/pihpsdr $ gdb ./pihpsdr
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./pihpsdr...done.
(gdb) r
Starting program: /home/gvj/Programs/pihpsdr/pihpsdr 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeddef700 (LWP 4736)]
Build: 2016-11-05 v1.0.0-beta-53-g3dbafa2-dirty
sysname: Linux
nodename: gvj-Mint
release: 3.19.0-32-generic
version: #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015
machine: x86_64
width=1920 height=1080
display_width=800 display_height=480
init
output_device: plughw:0,0 HDA NVidia
output_device: plughw:0,1 HDA NVidia
input_device: plughw:0,0 HDA NVidia
input_device: plughw:0,2 HDA NVidia
old_discovery
discover: looking for HPSDR devices on eth0
discover: bound to eth0
[New Thread 0x7fffdf6c4700 (LWP 4737)]
discover_receive_thread
discovered: received 60 bytes
discovery: found device=6 software_version=31 status=2 address=192.168.1.108 (00:1C:C0:A2:22:5D) on eth0
discovery: bytes read -1
discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
discovery: exiting discover_receive_thread
discover: exiting discover for eth0
[Thread 0x7fffdf6c4700 (LWP 4737) exited]
discovery found 1 devices
discovery: found device=6 software_version=31 status=2 address=192.168.1.108 (00:1C:C0:A2:22:5D) on eth0
new_discover: looking for HPSDR devices on eth0
new_discover: bound to eth0 192.168.1.50 255.255.252.0
[New Thread 0x7fffdf6c4700 (LWP 4738)]
new_discover: bytes read -1
new_discover: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
new_discover: exiting new_discover_receive_thread
new_discover: exiting discover for eth0
[Thread 0x7fffdf6c4700 (LWP 4738) exited]
new_discovery found 1 devices
discovery: found protocol=0 device=6 software_version=31 status=2 address=192.168.1.108 (00:1C:C0:A2:22:5D) on eth0
/home/gvj/Programs/pihpsdr/pihpsdr: found 1 devices.
0x65f460 protocol=0 name=Hermes Lite
start_cb: 0x65f460
selected radio=0x65f460 device=6
loadProperties: 00-1C-C0-A2-22-5D.props
wdsp_init: rx=0 pixels=800 protocol=0
OpenChannel 0 buffer_size=1024 fft_size=4096 sample_rate=48000 dspRate=48000 outputRate=48000
[New Thread 0x7fffdf6c4700 (LWP 4739)]
OpenChannel 8 buffer_size=1024 fft_size=4096 sample_rate=48000 dspRate=48000 outputRate=48000
[New Thread 0x7fffde879700 (LWP 4740)]
XCreateAnalyzer 0
SetAnalyzer channel=0 buffer_size=1024
SetAnalyzer channel=8 buffer_size=1024
SetRXAMode: rx=0 mode=0
SetTXAMode: tx=8 mode=0
SetAnalyzer channel=11 buffer_size=1024
old_protocol_init
audio_open_output: selected=0:plughw:0,0 HDA NVidia
audio_open_input: selected=0:plughw:0,0 HDA NVidia
[New Thread 0x7fffd73fa700 (LWP 4741)]
old_protocol starting receive thread: buffer_size=1024 output_buffer_size=1024
[New Thread 0x7fffd6bf9700 (LWP 4742)]
old_protocol_init: prime radio
old_protocol: receive_thread
FPGA firmware version: 3.1
vfo_height=60
vfo_init: width=400 height=60
vfo_init: set Frequency,Mode,Filter
setMode: mode=0 m=4
setMode: 0 mode=0
SetRXAMode: rx=0 mode=4
setMode: 8 mode=0
SetTXAMode: tx=8 mode=4
setFilter: 125 125
setFilter: filterLow=575 filterHigh=825
setFilter: fl=575.000000 fh=825.000000
menu_height=60
menu_init: width=200 height=60
input_device: plughw:0,0 HDA NVidia
input_device: plughw:0,2 HDA NVidia
output_device: plughw:0,0 HDA NVidia
output_device: plughw:0,1 HDA NVidia
meter_height=60
[New Thread 0x7fffd63f8700 (LWP 4743)]
meter_init: width=200 height=60
panadapter_height=120
waterfall_height=180
sliders_height=90
sliders_init: width=800 height=90
toolbar_height=30
toolbar_init: width=800 height=30 button_width=100
vfo_configure_event_cb: width=400 height=60
meter_configure_event_cb: width=200 height=60
panadapter_configure_event_cb: width:800 height:120
[New Thread 0x7fffd53f7700 (LWP 4744)]
[New Thread 0x7fffd4bf6700 (LWP 4745)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd6bf9700 (LWP 4742)]
process_ozy_input_buffer (buffer=<optimized out>) at old_protocol.c:500
500	        micinputbuffer[samples*2]=0.0;
(gdb) bt
#0  process_ozy_input_buffer (buffer=<optimized out>) at old_protocol.c:500
#1  0x00000000004143cb in receive_thread (arg=<optimized out>)
    at old_protocol.c:330
#2  0x00007ffff7378184 in start_thread (arg=0x7fffd6bf9700)
    at pthread_create.c:312
#3  0x00007ffff52ea37d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) q
A debugging session is active.

	Inferior 1 [process 4732] will be killed.

Quit anyway? (y or n) y

radi8, 2016-11-09:
I went back over the old builds in git as follows ...

beta-38 release (ALSA) ...................... OK
fixed install.sh to be latest version  .... OK
updated install.sh .............................. OK
old protocol source changes for local microphone .. No longer produces RF on MOX or TUNE but does not segfault.
Updated build document for ALSA .... Segfaults

This should help, 73 Graeme zl2apv

g0orx, 2016-11-09:
Can you update to the latest git version and try again.  Please note the 
latest version requires the updated wdsp on github as well.

-- John

radi8, 2016-11-09:
Everything updated and compiled 100% on wdsp and pihpsdr. It runs fine on Rx but still produces a segfault when operating MOX or TUNE. The window does not crash now but of course is locked up.

toolbar_height=30
toolbar_init: width=800 height=30 button_width=100
[New Thread 0x7fffd63f8700 (LWP 8649)]
vfo_configure_event_cb: width=400 height=60
meter_configure_event_cb: width=200 height=60
panadapter_configure_event_cb: width:800 height:120
[New Thread 0x7fffd53f7700 (LWP 8650)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd6bf9700 (LWP 8648)]
process_ozy_input_buffer (buffer=<optimized out>) at old_protocol.c:500
500	        micinputbuffer[samples*2]=0.0;
(gdb) bt
#0  process_ozy_input_buffer (buffer=<optimized out>) at old_protocol.c:500
#1  0x000000000041468b in receive_thread (arg=<optimized out>)
    at old_protocol.c:330
#2  0x00007ffff7378184 in start_thread (arg=0x7fffd6bf9700)
    at pthread_create.c:312
#3  0x00007ffff52ea37d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb)

g0orx, 2016-11-09:
> process_ozy_input_buffer (buffer=) at old_protocol.c:500
> #0 process_ozy_input_buffer (buffer=) at old_protocol.c:500

I just realized I was testing with the new Ethernet protocol.  I have 
pushed a fix now that should work with the old protocol.

-- John

radi8, 2016-11-09:
Thanks John,

It is a brilliant radio and in many ways more convenient to use than some of the powerhouse ones. The MOX and TUNE segfaults are gone and I can enjoy the improvements and enhancements you have made.

I noticed that I can no longer drive my Hermes-Lite to the full +20 dBm out on TX but of course with your radio still being in development I can not know if the previous versions which gave full output were giving too much drive for the Hermes radios and you have throttled it back or if you have inadvertently reduced it while developing.

Thanks again, Graeme zl2apv

g0orx, 2016-11-10:
Hi Graeme, 
I have made some changes to the driver to get a clean 100 watts out of the ANAN radios.
I have forgotten how the hermes lite controls the level. I will have a look to see what I can do.
-- John

radi8, 2016-11-10:
I thought that something like that would be the case and really appreciate your looking into this. I have been amazed at your responsiveness over all issues to date but don't put yourself under pressure on my account. I suspect that this radio will see a lot of use from Hermes-Lite users especially in its RaspberryPi stand alone form in the future. I'll close this MOX, TUNE issue as it is working perfectly now and leave the TX power item in your hands. Of course I am available for testing any time you request.
73, Graeme

closed 2016-11-10 by radi8