g0orx/pihpsdr #13: reserved identifier violation

elfring, 2017-01-30:
I would like to point out that identifiers like “[`_RADIO_H`](https://github.com/g0orx/pihpsdr/blob/17ab3548d7d6b9da1a141fc3e8b631cd16bacef9/radio.h#L20 "Update candidate")” and “[`_TOOLBAR_H`](https://github.com/g0orx/pihpsdr/blob/4102e268bd0ea5d4c59c58a09387e6f66aee6a48/toolbar.h#L20 "Another update candidate")” [do not fit](https://www.securecoding.cert.org/confluence/display/c/DCL37-C.+Do+not+declare+or+define+a+reserved+identifier#DCL37-C.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not use identifiers which are reserved for the compiler implementation.") to the expected naming convention of the C language standard.
Would you like to adjust your selection for unique names?

g0orx, 2017-02-18:
Development of the code is more important at the moment.

closed 2017-02-18 by g0orx

elfring, 2017-02-19:
How do you think about to avoid that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?")?

g0orx, 2017-02-19:
It is open source so feel free to fork a branch, make the changes and 
then request a merge,