g0orx/pihpsdr #14: Completion of error handling

elfring, 2017-01-30:
Would you like to add more error handling for return values from functions like the following?
- [fclose](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fclose.html "Close a stream.") ⇒ [saveProperties](https://github.com/g0orx/pihpsdr/blob/1b95bf1a08e5f860922f09a307aac7955ea5a3f0/property.c#L62)
- [fgets](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fgets.html "Get a string from a stream.") ⇒ [check_update](https://github.com/g0orx/pihpsdr/blob/8d3cf8d70229196d179cf49b9e22c308d5b663cd/update.c#L13)

g0orx, 2017-02-18:
Development of the code is more important at the moment.

closed 2017-02-18 by g0orx

elfring, 2017-02-19:
I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Unchecked return values") a bit more.
Would you like to detect every error situation as early as possible?