g0orx/pihpsdr #16: Odroid-X2 Ubuntu 14.04 32bit

KC7NOA, 2017-02-18:
I get an error building .. and should be the same OS image as the C2/C2 ..MATE ..

kc7noaodroid@odroid:~/Downloads/pihpsdr$ make
rm -f version.o
gcc -g -Wno-deprecated-declarations -D odroid         -D SHORT_FRAMES -D GIT_DATE='"2017-02-18"' -D GIT_VERSION='"v1.1.1-beta"'  -O3 `pkg-config --cflags gtk+-3.0` -c -o audio.o audio.c
Package gtk+-3.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gtk+-3.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gtk+-3.0' found
audio.c:21:21: fatal error: gtk/gtk.h: No such file or directory
compilation terminated.
Makefile:366: recipe for target 'audio.o' failed
make: *** [audio.o] Error 1

g0orx, 2017-02-18:
You have not installed the gtk development package.   Have you looked at the build document at:

https://github.com/g0orx/pihpsdr/blob/master/release/documentation/pihpsdr-build.pdf

closed 2017-02-18 by g0orx

KC7NOA, 2017-02-18:
oh .. comeon ... my eyes must be getting old ....

I read that three time in this last month ... prolly assumed that it was already part of the Pi/Odroid image ..

https://github.com/g0orx/pihpsdr/blob/master/release/documentation/pihpsdr-build.pdf

[https://assets-cdn.github.com/images/spinners/octocat-spinner-128.gif]<https://github.com/g0orx/pihpsdr/blob/master/release/documentation/pihpsdr-build.pdf>

pihpsdr/pihpsdr-build.pdf at master · g0orx/pihpsdr · GitHub<https://github.com/g0orx/pihpsdr/blob/master/release/documentation/pihpsdr-build.pdf>
github.com
pihpsdr - Raspberry Pi 2 (and Odroid C1+) standalone code for HPSDR (new Ethernet protocol)