g0orx/pihpsdr #17: install issue ...

KC7NOA, 2017-02-19:
im not sure if i caused this because i built and ran sudo make install from /Downloads/pihpsdr/

kc7noaodroid@odroid:~/Downloads/pihpsdr$ sudo make install
[sudo] password for kc7noaodroid: 
cp pihpsdr ../pihpsdr
cp: 'pihpsdr' and '../pihpsdr/pihpsdr' are the same file
Makefile:374: recipe for target 'install' failed
make: *** [install] Error 1
kc7noaodroid@odroid:~/Downloads/pihpsdr$

g0orx, 2017-02-19:
Please read the install instructions. Note that the install instructions are for the Raspberry Pi but should work on other platforms.

https://github.com/g0orx/pihpsdr/blob/master/release/documentation/pihpsdr-install.pdf

Note that the install.sh script is not run as sudo and the pihpsdr directory is expected to be in the home directory of the user.

KC7NOA, 2017-02-20:
hummm .... the install script is located in 
/pihpsdr/release/pihpsdr/

I didn't use the wget -N simply because the PDF is located in the release folder ... 
its a chicken or the egg issue. 

I used git clone ... this is a git repository after all ...

Those instructions might be better in the README.md or a wget to the PDF directly in the README.md

Now .. when i ran the install.sh .... the only thing i seen that i didn't already have was libfftw3-3 and that pulled in alot of others stuff ... including xscreensaver that i don't really want --

Probably should put in the PDF and README.md that pihpsdr should be extracted into home pi(USER) folder .... to reduce confusion.

KC7NOA, 2017-02-20:
So after moving the folder to HOME
runnung install.sh from pihpsdr/release/pihpsdr/

these returned info probably dose not mean much and can be ignored.

setting up for USB OZY device
cp: cannot stat '90-ozy.rules': No such file or directory
creating desktop icon
cp: cannot stat 'pihpsdr.desktop': No such file or directory
cp: cannot stat 'pihpsdr.desktop': No such file or directory

g0orx, 2017-02-20:
This is open source, so feel free to fork the code and make changes and 
request a merge and I can review the changes and merge if they make 
sense. Note also that the current release is a Beta release so expect 
there to be some issues.

closed 2017-04-07 by g0orx