g0orx/pihpsdr #24: /home/kc7noaa/Downloads/pihpsdr/receiver.c:805: undefined reference to `RXASetNC'

KC7NOA, 2017-03-09:
how soon is too soon to post an error

receiver.o: In function `create_receiver':
/home/kc7noaa/Downloads/pihpsdr/receiver.c:805: undefined reference to `RXASetNC'
/home/kc7noaa/Downloads/pihpsdr/receiver.c:806: undefined reference to `RXASetMP'
transmitter.o: In function `create_transmitter':
/home/kc7noaa/Downloads/pihpsdr/transmitter.c:420: undefined reference to `TXASetNC'
/home/kc7noaa/Downloads/pihpsdr/transmitter.c:421: undefined reference to `TXASetMP'
collect2: error: ld returned 1 exit status
Makefile:363: recipe for target 'pihpsdr' failed
make: *** [pihpsdr] Error 1

g0orx, 2017-03-09:
Requires updated wdsp.

-- John

KC7NOA, 2017-03-09:
OK ... good to go forward

closed 2017-03-09 by KC7NOA