g0orx/pihpsdr #30: Problems using Beaglebone Black (with 7" display)

dm6tt, 2017-04-06:
Hi folks,
I switched the display of my beaglebone to a 7" (1024x600.)
Compiling the code was no problem, but when I start, I get full_rx_buffer errors and the waterfall / spectrum of the program is not displayed.
The Button/sliders can be seen, but one core is completely busy.

I use a debian system: $ uname -a
Linux beaglebone 3.8.13 #1 SMP Sat Aug 9 11:31:19 CST 2014 armv7l GNU/Linux

The SDR is a redpitaya. This works fine with the same pihpsdr version compiled on x86 and a raspy3.

Here is the console log from the beaglebone:

**Build: 2017-03-10 v1.1.3-beta
sysname: Linux
nodename: beaglebone
release: 3.8.13
version: #1 SMP Sat Aug 9 11:31:19 CST 2014
machine: armv7l
width=2560 height=1024
display_width=800 display_height=480
create top level window
create grid
add grid
create image
add image to grid
create pi label
add pi label to grid
create build label
add build label to grid
create status
add status to grid
init
audio_get_cards
input_device: plughw:0,0 DA830 EVM
output_device: plughw:0,0 DA830 EVM
discovery
old_discovery
discover: looking for HPSDR devices on eth0
discover: bound to eth0
discover_receive_thread
discovered: received 60 bytes
discovery: found device=0 software_version=21 status=3 address=192.168.10.70 (00:26:32:F0:12:D8) on eth0
discovery: bytes read -1
discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
discovery: exiting discover_receive_thread
discover: exiting discover for eth0
discovery found 1 devices
discovery: found device=0 software_version=21 status=3 address=192.168.10.70 (00:26:32:F0:12:D8) on eth0
new_discover: looking for HPSDR devices on eth0
new_discover: bound to eth0 192.168.10.167 255.255.255.0
new_disovery: thread_id=0x68ec0
new_discover: bytes read -1
new_discover: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
new_discover: exiting new_discover_receive_thread
new_discover: exiting discover for eth0
new_discovery found 1 devices
discovery: found protocol=0 device=0 software_version=21 status=3 address=192.168.10.70 (00:26:32:F0:12:D8) on eth0
discovery: found 1 devices
0x42344 protocol=0 name=Metis
showing device dialog
discovery
old_discovery
discover: looking for HPSDR devices on eth0
discover: bound to eth0
discover_receive_thread
discovered: received 60 bytes
discovery: found device=0 software_version=21 status=3 address=192.168.10.70 (00:26:32:F0:12:D8) on eth0
discovery: bytes read -1
discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
discovery: exiting discover_receive_thread
[... 256 lines cut ...]
RIGCTL: Bind done on port 19092
RIGCTL: Waiting for incoming connections...
RIGCTL: Socket created
RIGCTL: Bind done on port 19090
RIGCTL: Waiting for incoming connections...
RIGCTL: Socket created
RIGCTL: Bind done on port 19091
RIGCTL: Waiting for incoming connections...
full_rx_buffer: id=0 fexchange0: error=-2
full_rx_buffer: id=1 fexchange0: error=-2
full_rx_buffer: id=0 fexchange0: error=-2
full_rx_buffer: id=1 fexchange0: error=-2
full_rx_buffer: id=0 fexchange0: error=-2
full_rx_buffer: id=1 fexchange0: error=-2
full_rx_buffer: id=0 fexchange0: error=-2
full_rx_buffer: id=1 fexchange0: error=-2
...
...**

I also tried to forward the display to my x86 machine, to exclude problems with the 7" display. The same effect there, only the sliders and menus, without waterfall/sprectrum is shown. The error messages are the same.

What can be the reason?

Thanks
73, Marcus, DM6TT

g0orx, 2017-04-06:
I looks like the board just does not have enough CPU to run the 
software. As I do not have one I am unable to test unless someone likes 
to send me one.  You could try just one receiver to see if can handle that.

-- John g0orx/n6lyt

dm6tt, 2017-04-06:
Thanks John for your reply.
I tried one receiver, without a better result.
I started the programm with strace and found:

Open connection for the X-Display (here the ssh forward on 6010:
**socket(PF_INET, SOCK_STREAM|SOCK_CLOEXEC, IPPROTO_TCP) = 6
setsockopt(6, SOL_TCP, TCP_NODELAY, [1], 4) = 0
setsockopt(6, SOL_SOCKET, SO_KEEPALIVE, [1], 4) = 0
connect(6, {sa_family=AF_INET, sin_port=htons(6010), sin_addr=inet_addr("127.0.0.1")}, 16) = 0**

and then very often timeouts reading from that socket:
**writev(6, [{"l\0\v\0\0\0\22\0\20\0\0\0", 12}, {"", 0}, {"MIT-MAGIC-COOKIE-1", 18}, {"\0\0", 2}, {"\274]\222c\216\303+\252\231$?a\207cQ\237", 16}, {"", 0}], 6) = 48
recv(6, 0x71540, 8, 0)                  = -1 EAGAIN (Resource temporarily unavailable)
...**

I just checked the CPU. It seems it runs with a Cortex-A7 with one core. So propably the board is really too slow.

I will go back to my rasperry3.

Thanks

73, Marcus

closed 2017-04-07 by g0orx