g0orx/pihpsdr #33: x86 version

W0FNQ, 2017-05-19:
I've been trying to run pihpsdr on my desktop 64bit mint machine.  It compiles and runs except when I try to execute the MENU RX button, I get the following segmentation fault as identified in GDB:

Thread 1 "pihpsdr" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/x86_64/strlen.S:106
106	../sysdeps/x86_64/strlen.S: No such file or directory.
and bt gives the following:

(gdb) bt
#0  strlen () at ../sysdeps/x86_64/strlen.S:106
#1  0x00007ffff54544e3 in g_strdup ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff5736e5c in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#3  0x00007ffff571415b in g_object_new_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#4  0x00007ffff5714521 in g_object_new ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#5  0x00007ffff69a5c97 in gtk_radio_button_new_with_label ()
   from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#6  0x000000000040ce74 in rx_menu (parent=<optimized out>) at rx_menu.c:306
#7  0x000000000040b632 in rx_cb (widget=<optimized out>, 
    event=<optimized out>, data=<optimized out>) at new_menu.c:102
#8  0x00007ffff6947fac in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#9  0x00007ffff570cfa5 in g_closure_invoke ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007ffff571efc1 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007ffff57277f9 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007ffff572808f in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#13 0x00007ffff6a85c3c in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#14 0x00007ffff69453be in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#15 0x00007ffff69471bc in gtk_main_do_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#16 0x00007ffff64b4d92 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#17 0x00007ffff5436197 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff54363f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007ffff543649c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007ffff59fdd7d in g_application_run ()
   from /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#21 0x000000000040749f in main (argc=1, argv=0x7fffffffe058) at main.c:266

g0orx, 2017-05-20:
Can you send the debug output. The problem seems to be in the display of local audio output devices.  Can you also let me know what Linux distribution you are running.  I do not see any problem when running on an i5 system using Ubuntu 16.10.
-- John

W0FNQ, 2017-05-20:
John,
I am running LInux Mint 18.1 64bit
Cinnamon Version 3.27
Linux Kernel 4.4.0-53-generic
Processor Intel Core I5-3750 CPU 3.4GHz x 4
Memory 15.6 GB
Graphic Card NVidia GTX660
Jack (W0FNQ)

I am not sure if this is what you need, let me know if something else>
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/home/jack/pihpsdr/pihpsdr'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  strlen () at ../sysdeps/x86_64/strlen.S:106
106	../sysdeps/x86_64/strlen.S: No such file or directory.
[Current thread is 1 (Thread 0x7ffff7f25a40 (LWP 6792))]
(gdb) bt
#0  strlen () at ../sysdeps/x86_64/strlen.S:106
#1  0x00007ffff54544e3 in g_strdup () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff5736e5c in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#3  0x00007ffff571415b in g_object_new_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#4  0x00007ffff5714521 in g_object_new () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#5  0x00007ffff69a5c97 in gtk_radio_button_new_with_label () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#6  0x000000000040ce74 in rx_menu (parent=<optimized out>) at rx_menu.c:306
#7  0x000000000040b632 in rx_cb (widget=<optimized out>, event=<optimized out>, data=<optimized out>) at new_menu.c:102
#8  0x00007ffff6947fac in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#9  0x00007ffff570cfa5 in g_closure_invoke () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007ffff571efc1 in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007ffff57277f9 in g_signal_emit_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007ffff572808f in g_signal_emit () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007ffff6a85c3c in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#14 0x00007ffff69453be in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#15 0x00007ffff69471bc in gtk_main_do_event () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#16 0x00007ffff64b4d92 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#17 0x00007ffff5436197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff54363f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007ffff543649c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007ffff59fdd7d in g_application_run () from /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#21 0x000000000040749f in main (argc=1, argv=0x7fffffffe058) at main.c:266
(gdb) bt full
#0  strlen () at ../sysdeps/x86_64/strlen.S:106
No locals.
#1  0x00007ffff54544e3 in g_strdup () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#2  0x00007ffff5736e5c in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#3  0x00007ffff571415b in g_object_new_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#4  0x00007ffff5714521 in g_object_new () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#5  0x00007ffff69a5c97 in gtk_radio_button_new_with_label () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
No symbol table info available.
#6  0x000000000040ce74 in rx_menu (parent=<optimized out>) at rx_menu.c:306
        output = <optimized out>
        local_audio_b = <optimized out>
        stereo_b = <optimized out>
        left_b = <optimized out>
        right_b = <optimized out>
        label = "RX 1 VFO B\000\000\000\000\000\000\300gz\000\000\000\000\000\310gz\000\000\000\000"
        adc_b = <optimized out>
        i = 20
        color = {red = 1, green = 1, blue = 1, alpha = 1}
        content = 0x7aa7a0
---Type <return> to continue, or q <return> to quit---
        grid = 0x7aade0
        close_b = <optimized out>
        rx_label = <optimized out>
        x = <optimized out>
        n_adc = 1
        row = 21
        mute_audio_b = <optimized out>
#7  0x000000000040b632 in rx_cb (widget=<optimized out>, event=<optimized out>, data=<optimized out>) at new_menu.c:102
No locals.
#8  0x00007ffff6947fac in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
No symbol table info available.
#9  0x00007ffff570cfa5 in g_closure_invoke () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#10 0x00007ffff571efc1 in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#11 0x00007ffff57277f9 in g_signal_emit_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#12 0x00007ffff572808f in g_signal_emit () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#13 0x00007ffff6a85c3c in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
No symbol table info available.
#14 0x00007ffff69453be in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
No symbol table info available.
---Type <return> to continue, or q <return> to quit---
#15 0x00007ffff69471bc in gtk_main_do_event () from /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
No symbol table info available.
#16 0x00007ffff64b4d92 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-3.so.0
No symbol table info available.
#17 0x00007ffff5436197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#18 0x00007ffff54363f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#19 0x00007ffff543649c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#20 0x00007ffff59fdd7d in g_application_run () from /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
No symbol table info available.
#21 0x000000000040749f in main (argc=1, argv=0x7fffffffe058) at main.c:266
        pihpsdr = 0x6701a0
        status = <optimized out>

W0FNQ, 2017-05-22:
John I found the problem.  When Mint installs libgtk-3-dev it leaves it in a directory inside of /usr/include with the name gdk-3.0.  A symbolic link  /usr/include/gdk --> /usr/include/gtk-3.0/gdk solved the issue.  When I open the project in Eclipse it was apparent.  I should have done than in the first place rather than bothering you.  Thanks and Sorry.  73 Jack (W0FNQ)