g0orx/pihpsdr #35: PSK Compile Error

frohro, 2017-06-17:
psk_waterfall.c: In function ‘psk_waterfall_update’:
psk_waterfall.c:203:20: error: ‘data’ undeclared (first use in this function)
             sample=data[i];
                    ^~~~
psk_waterfall.c:203:20: note: each undeclared identifier is reported only once for each function it appears in
Makefile:367: recipe for target 'psk_waterfall.o' failed
make: *** [psk_waterfall.o] Error 1

Analias, 2017-06-25:
I'm seeing the same error. I don't see where the reference to 'data' should be resolved to.

What's funny is every reference to data else were in the file is an argument to each function that is never used. Go figure.

frohro, 2017-06-25:
John is actively working on adding features and is aware of the compile error.  It will be fixed as soon as the new stuff is ready.

holgerschurig, 2018-03-31:
Well, this half-a-year-old bug hasn't been fixed by anybody. So I guess no one really cares about bug reports.

I can at least the this project to compile if I add the line `float *data = rx->pixel_samples` just before the for loop, two lines above the reported error. As I don't currently have a HPSDR capable hardware I cannot test if this is then working. But now the code in psk_waterfall.c is similar to waterfall, so maybe this is enough.