g0orx/pihpsdr #39: V1.2 CW fault(?)

df5sf, 2017-11-12:
Hi John,

My settings:  CW-Keyer Mode B, Drive=100, Tune use driver = 100%, CWL and RedPitaya
I use no VOX/PTT or something like this.
After key-down (dot or dash), it starts not every time to send HF. It is like gambling.

My version: build 2017-11-12 v1.2 installed from .tar file

g0orx, 2017-11-12:
I do not have a RedPitaya so cannot test, but it is working OK with my ANAN-100D, ANAN-800DLE and Hermes board.

ka6s, 2017-11-16:
Hi,

I have a Red Pitaya - and haven't been able to duplicate your issue.  On the other hand - I likely have a much older version of Pavel's code - so it's hard to draw conclusions.  One thing I wanted to suggest was for you to wipe the .props file completely - and start from ground zero.  We've seen things work weird sometimes between versions of the software - and the .props file has been the culprit.  You can contact me at ka6s@arrl.org if you want to follow up..

dl1ycf, 2018-06-08:
Your RedPitaya ignores DriveLevel change commands while doing CW. I have fixed this in my
pull request, you can look at the code in old_protocol.c. Is is now activated when choosing
the new STEMLAB_FIX_OPTION in the Makefile.
Yours DL1YCF.