g0orx/pihpsdr #42: V1.2 pin assotiation for encoders corrupted

mompelix, 2017-11-20:
Hi, thank you for this wonderful program. I updated from a former verion to 1.2 and deleted the props first. so i started from scratch. But now the encoder pins semms to be corrupted: The VFO knob doesn't work any more. I have 3 other Encoders. they do crazy thing when i operate then. Any idea?

g0orx, 2017-11-20:
The updated install document says to rename or remove the existing 
pihpsdr directory before extracting the tar file.  The new code uses 
wiringPi and it's pin numbering so that it will also work with the ASUS 
Tinker Board. There are a few other changes to the install.sh script as 
well so that it is not specific for the Raspberry Pi.

-- John g0orx/n6lyt

mompelix, 2017-11-20:
Hello John,
yeah, that helped a lot.
Thanks
Herko
DK3HU

(#42)
 
 
 

 
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-b52498112777.png","action":{"name":"Open 
in 
GitHub","url":"https://github.com/g0orx/pihpsdr"}},"updates":{"snippets":[{"icon":"PERSON","message":"@g0orx 
in #42: The updated install document says to rename or remove the existing 
\npihpsdr directory before extracting the tar file.  The new code uses 
\nwiringPi and it's pin numbering so that it will also work with the ASUS 
\nTinker Board. There are a few other changes to the install.sh script as 
\nwell so that it is not specific for the Raspberry Pi.\n\n-- John 
g0orx/n6lyt\n\n\nOn 20/11/17 21:22, mompelix wrote:\n\u003e\n\u003e Hi, 
thank you for this wonderful program. I updated from a former \n\u003e 
\n\u003e But now the encoder pins semms to be corrupted: The VFO knob 
doesn't \n\u003e work any more. I have 3 other Encoders. they do crazy 
thing when i \n\u003e operate then. Any idea?\n\u003e\n\u003e —\n\u003e You 
are receiving this because you are subscribed to this thread.\n\u003e Reply 
to this email directly, view it on GitHub \n\u003e 
\n\u003e 
Issue","url":"https://github.com/g0orx/pihpsdr/issues/42#issuecomment-345852595"}}}

closed 2019-11-18 by mompelix