g0orx/pihpsdr #46: Bluetooth audio

W9JSW, 2018-01-06:
How can I use Bluetooth audio? Paired Anker Soundcube. See no entry on a play -l or on RX config to select it, only USB audio dongle. This may be a stretch issue as well, in that I can select the paired device but get no default audio from it.

W9JSW, 2018-01-08:
Upon further investigation, pairing and selecting the bluetooth device creates a virtual PCM named bluealsa. It has an accessible device id but does not get listed with aplay -L.

holgerschurig, 2018-03-31:
This is IMHO not a piHPSDR question, so I'd suggest that this issue get closed.

piHPSDR uses ALSA. **Any** program that uses ALSA can be made to use PulseAudio, e.g. with `padsp` or also via alsa configuration files. And with PulseAudio you can route audio to wherever you please, independent of the program, e.g. with pavucontrol. 

So what you lack currently is not a option in piHPSDR, but knowledge on how to tame the ALSA/PulseAudio beast :-)

I'm sure that with this pointers you can google for a solution. Maybe this is a starting point: https://www.howtoforge.com/tutorial/how-to-send-sound-through-bluetooth-on-linux/ or, ham radio related, https://groups.io/g/linuxham/topic/connect_pihpsdr_to_fldigi/4215601?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,4215601

W9JSW, 2018-03-31:
I disagree. Yes, you can use pulseaudio to resolve this issue and 
thereby add an order of magnitude of complexity to the average user to 
glean how to figure it out. The target audience, IMHO, is not linux 
savvy. It would seem that if I have my handy bluetooth headset, and I 
pair it to the OS, I should then see an entry in the RX and TX screens 
allowing me to select and use it. Jumping off into synaptic, installing 
a slew of additional software, and hopefully not messing it up too badly 
is well beyond the capability of the average Anan user.

Also, the Alsa implementation is not that solid. For example, I have an 
alsa loopback device configured for fldigi. Most of the time I have 
c-media selected for my sound for SSB. If I reboot and fail to 
re-establish the loopback device prior to launch, pihpsdr will crash 
when I attempt to reselect audio. The audio support needs more work...

John W9JSW