g0orx/pihpsdr #63: STEMLAB compile fails

dcmk1mr2, 2018-07-22:
In the instructions in the C25-SDR pihpsdr recommends installing avahi and cURL for STEMLAB.  "sudo apt-get install avahi" results in no package found. 

Uncommenting the STEMLAB_DISCOVERY=STEMLAB_DISCOVERY option in the pihpsdr Makefile causes a compilation error.

dl1ycf, 2018-08-03:
I now also have a Raspberry Pi and compiled pihpsdr for my Hamlab (C25). Since I also did not know how to install avahi, I compiled it with -D NO_AVAHI than you need not avahi (but you need libcurl).
To get this to work, you must indicate the ip address (e.g. 192.168.1.100) of your RedPitaya
in a while with name  $HOME/.rp.inet (it is best to set the RedPitaty inet address to a fixed
value in this setup).

However, as far as I know there is a working avahi lib out there.

To get rid of compilation errors, you have to "make clean" after changing anything in the Makefile.

SO FOR THE TIME BEING
==================

The latest version with the unified STEMLAB support for Macintosh/Linux is on my github account,
please try this and in the Makefile simply replace

STEMLAB_OPTIONS=-D STEMLAB_DISCOVERY \
  `pkg-config --cflags avahi-gobject` \
  `pkg-config --cflags libcurl`
STEMLAB_LIBS=`pkg-config --libs avahi-gobject` `pkg-config --libs libcurl`

with 

STEMLAB_OPTIONS=-D STEMLAB_DISCOVERY -D NO_AVAHI `pkg-config --cflags libcurl`
STEMLAB_LIBS=`pkg-config --libs libcurl`

which is the version from Makefile.mac. I have just tried this on my Raspberry PI and it works.

dcmk1mr2, 2018-08-04:
I tried your suggestion and still get 
stemlab_discovery.c:232:37: fatal error: avahi-gobject/ga-client.h: No such file or directory
 #include <avahi-gobject/ga-client.h>