g0orx/pihpsdr #74: pihpsdr

holsteinjohan, 2019-08-04:
Hi all,

I use the radioberry in combination with pihpsdr this work very nice, however I like to use it as a standalone unit in cobination with a 7" touchscreen. I prever this ....  start the rpi and final I see the pihpsdr, the start button for choosing a network must be removed in my case.
Hope someone can help with that.....
Best regards,
Johan Holstein

dl1ycf, 2019-08-16:
The program is designed to run with many radios. So we *need* the START button in case there are two radios on the network, where you have to choose from. So what prevents you from pushing the "START" button?

holsteinjohan, 2019-12-26:
There is in my case no need to use the start button, where in the software I can by-pass this.
Best,
Johan

dl1ycf, 2019-12-26:
I am afraid I have not understood the comment. Please note, the START button is absolutely necessary since there might be more than one radio connected and you want to choose. Moreover, it can be necessary to configure the GPIO, "import" MIDI description files, etc. before actually starting the radio.

If you want to create a local version that automatically starts the first radio that has been found,
you can do so in a local version, but I do not see at the moment that this should be pihpsdr's default behaviour.

holsteinjohan, 2019-12-26:
Hi,

The start button is totally unimportant for me when using a standalone installation. The question was not to implement this in future version but how to change the code at my place local. Anyway if it seems that difficult than I will Digg into the code myself.

https://sites.google.com/view/pd0lew/projects/radioberry

Best regards,

Johan Holstein

dl1ycf, 2019-12-26:
not difficult.

It is sufficient to set the variable „radio“ to &discovered[i] where i is the first device found,
and then call start_radio() from discovery.c

The only thing one must try is whether it can be called directly (e.g. after the line reading

fprintf(stderr,"showing device dialog\n“);

of whether one must wrap this into a function eventually called by g_idle_add().