g0orx/pihpsdr #77: Pihpsdr compilation

UN7RX, 2019-10-23:
Good afternoon, dear colleagues!
I apologize for the bad English.
I program a little in VB, C #, assembler. But I have never worked with Linux and with the Raspberri Pi. Please tell me what I need from the software to compile pihpsdr from open source? I want to try changing the program interface a bit so that it becomes more convenient and bright, such as RaSDR. Font sizes, layout of controls. This is a great program, many thanks to G0ORX for this wonderful product!

dl1ycf, 2019-11-21:
There is a step-by-step instruction (which I have written) now in the repository, look at
release/documentation/RaspPi+Tinker-piHPSDR-CompileFromSources.pdf

It also applies to most Linux boxes. If you want to compile on MacOS, things are slightly different, but there is also a MacOS README.

DL1YCF

g4fkk, 2020-03-25:
Dear DL1YCF,
I followed your instructions to the letter but when it came to compiling piHPSDR I got a number of fatal errors, all along the lines of...
make: *** [Makefile 189: discovery.o] Error 1
In file included from radio.h:25,
                         from audio.c:39:
discovered.h:25:10: fatal error: SoapySDR/Device.h: No such file or directory
 #include <SoapySDR/Device.h>
                ^~~~~~~~~~~~~~~
compilation terminated.
make: [Makefile:189: audio.o] Error 1

As a complete Linux/Raspian novice most of this means very little to me other than to suggest the compiler cannot find SoapySDR.  Any suggestions as to what I should do/try next gratefully received :)

73 Martin - G4FKK

g0orx, 2020-03-25:
Hi Martin,

The current Makefile include SoapySDR support . If you are not going to 
be using a device supported by SoapySDR such as a LimeSDR, PlutoSDR or 
RTLSDR then you can disable support for SoapySDR by commenting out the 
SoapySDR define in the Makefile by putting a # at the beginning of the line:

#SOAPYSDR_INCLUDE=SOAPYSDR

If you want to compile to support SoapySDR then you will need to install 
SoapySDR and the device drivers for the devices you want to use.

Updated documentation is in process of being put together.

-- John g0orx/n6lyt

g4fkk, 2020-03-25:
Hi John,

Many thanks for your prompt reply and your answer to my, in retrospect, rather obvious question!  I was trying to set things up very early this morning which is never a good idea hi.  At about 02:00 I thought "oh well, nothing to lose" so I installed "useful-looking-SoapySDR-stuff" via the Pi's GUI.  To my surprise, this worked; piHPSDR compiled and appears to be working as expected.

I've built an SDR transceiver using a Red Pitaya and have been using the pre-compiled piHPSDR on it for about a year very successfully (see my QRZ page for a picture).  Recently though, piHPSDR has decided not to show any zeroes in the frequency, step size and RIT displays.  I thought I'd re-download the pihpsdr.tar file and re ./install it but I appear not to be able to find that file on github.  Hence my attempt at self compilation (helps with self isolation I suppose).

I do have a Lime Mini which I use for DATV.  I'll try plugging that in next :)

Thanks again for your reply and for all your astonishing work on this project!!

73 Martin - G4FKK

dl1ycf, 2020-03-25:
If you read the instructions carefully, you will find that
after down-loading everything, you should adjust the Makefile.

In your case, it is quite obvious that you have *not* commented out
the line 

SOAPYSDR_INCLUDE=SOAPYSDR

which should read in your case

#SOAPYSDR_INCLUDE=SOAPYSDR

Of course, if you *need* support for SOAPY-SDR then you have to install 
recent versions of libsoaopysdr on your machine. According to John, this
is not easy because the version of soapy that is currently in the „packages“
is out-dated and you have to get the source code of a current soapy version
and compile/install it yourself.

dl1ycf, 2020-03-25:
The good news for RedPitaya is, that pihpsdr can
start your SDR application on the RedPit
so you need not connect via a browser, start the SDR
and (only) then start pihpsdr

g4fkk, 2020-03-25:
Hello DL1YCF,

I did read the instructions but, due to my lack of knowledge, I didn't know I had to comment out the SoapySDR references :)  As mentioned above, I have now managed to get it to compile by downloading the "packages" but, from your comments, I'm not expecting my Lime Mini to function any time soon.

I've been using the pre-compiled version of piHPSDR with my Red Pitaya for a long time now and, as you say, it just works with no need to connect to a WAN or make use of a browser.  Fabulous!

Thanks for your reply and especially for your excellent step-by-step instructions - very helpful for a Linux Dummy like me!

73 Martin - G4FKK

dl1ycf, 2020-03-25:
Well my main development machine is a Macintosh, and there I can install recent
SOAPY libs from the standard archives. I think the minimum version of libsoapy
is 0.7, better 0.8. A while ago I tried to install these libs on the RaspPi
via „apt-get“ and got some version 0.52 and that is definitely too old.

John is distributing the SOAPY libs with his binary version, but obviously
only those you need to run an executable, some more is required to 
compile it (that is, lib-devel rather than lib). You may ask him whether
he can provide a „devel“ tar-ball as well.

g0orx, 2020-03-25:
I have pushed an updated version of 
RaspPi+Tinker-piHPSDR-CompileFromSources.pdf to github. This now has an 
Appendix with instructions on how to compile SoapySDR and modules for 
LimeSDR, PlutoSDR and RTLSDR.

-- John

g4fkk, 2020-03-26:
Hi John,

I followed your modified instructions on a new SD card and everything appears to be working as expected with the Red Pitaya and pi 3 B+.  I've not tried the Lime mini yet (it is nearly 02:00 :))

A couple of notes which may be of interest...

a) The first time after firing up pihpsdr, switching Band to 80m actually switches to a 60m frequency.  I then have to enter an 80m frequency using Freq and then swap VFOs a couple of times to get the correct LPF to switch in.  I've taken to 'storing' an 80m frequency in VFO B and copying it to VFO A.

b) It may be worth adding - **cat /sys/module/kernel/parameters/consoleblank** - to your instructions to stop the 7" display going blank every few minutes.  I found this via Google and it may not be a good thing.  It seems to do the trick though.

Many thanks again to you and dl1ycf for all your help,

73 Martin - G4FKK

g0orx, 2020-03-26:
I suspect the band change problem is because you have an old props file for
the radio. Try deleting the props file and see if that fixes your band
change problem.

Because of all the changes there have been a lot of changes to the props
file.

I will add some information on stopping the screen blanking.

-- John

dish-eme-sdr, 2020-03-26:
In the section (near the end) titled:
Download, compile and install SoapyRTLSDR

Had to change:
git clone https://github.com/pathosware/SoapyRTLSDR

To the following:
git clone https://github.com/pothosware/SoapyRTLSDR

Typo ???

-=<DISH>=-

dl1ycf, 2020-03-26:
I have updated the documentation, perhaps look at
the version in 

github.com/dl1ycf/pihpsdr

that is not yet merged into the master repo.

Did you succeed in compiling this? I had undefined references
to __atomic_load_8 and others so I have to edit CMakeLists.txt
to add „-latomic“:

diff --git a/CMakeLists.txt b/CMakeLists.txt
index b8ffb00..142de90 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -34,7 +34,7 @@ if(CMAKE_COMPILER_IS_GNUCXX)
    endif()

    #Thread support enabled (not the same as -lpthread)
-    list(APPEND RTLSDR_LIBRARIES -pthread)
+    list(APPEND RTLSDR_LIBRARIES -pthread -latomic)

This is not yet in the documentation since I want to check with others
if this problem also appears.

g4fkk, 2020-03-26:
Latest news from G4FKK...

Following John's updated instructions I managed to compile everything onto a new SD card using a Pi 3B+.  This worked with the Red Pitaya.  I tried the Lime Mini but the Pi didn't seem to have enough grunt.  The display was very slow and no audio.

I got another clean SD card and this time, following John's modified instructions, compiled it all on a Pi 4 Model B.  Again, everything compiled with no errors and, once again, worked fine with the Red Pitaya.  Trying the Lime Mini looked as if everything was working.  Receiver noise sounded as if plugging in an aerial should produce some signals.  No signals received however.  I tried 144MHz and 432MHz where I have signal generators available but nothing at all.

The Lime Mini works fine on these bands using SDR Console so I don't think it's faulty.  I wonder though, if there are some settings I don't understand?  SoapySDR compiled fine as did LimeSuite.  The Lime Mini's LED flashed green as it does using SDR Console and the BATC Portsdown software.

Let me know if there's any info I could usefully provide.

By the way, I got the same 80m/60m issue as reported before and both SD cards were brand new.  I'm assuming John, that the props file you suggested deleting resides on the SD card or does it live somewhere on the Red Pitaya?

73 Martin - G4FKK

UN7RX, 2020-04-24:
Dear colleagues, unfortunately I'm not good at Linux and C yet, tell me please, is there such a programming environment in which you can change the pihpsdr GUI using visual programming?
PiHPSDR is the best program for creating a transceiver in a classic format. Especially with the creation of the PiHPSDR-Controller-MKII. Of course, appearance is a matter of taste. But I would like to slightly change the font size for the frequency indicators, the size of the S meter, to make the display colors brighter. In general, so that everything is not so ascetic)) But in order to do this purely through code, you need to know C at least half of how John knows him)))

dl1ycf, 2020-04-24:
I just do not see what you mean. pihpsdr uses the GTK GUI, and you can adapt it somewhat to your "feel" using a different theme. But many things are hard-coded, even the screen size. So do not do much of adaption.

While in principle one can take out the "engine" of the car and put it into another one (another GUI), I see no point therein.

dl1ycf, 2020-04-24:
Well, now I see: to make the S meter larger, you have to increase the amount of space used by
the "VFO" bar. Thinking about this, this is a no-go although easy for a C programmer. The program is designed to run on a small (7 inch) screen, so space is scarce.

UN7RX, 2020-04-25:
> But many things are hard-coded, even the screen size. So do not do much of adaption.

This is not a problem. You do not even need to program for this. You just need to remove the constants. Here it is shown how this can be done.  https://charly25-sdr.github.io/software/pihpsdr

> The program is designed to run on a small (7 inch) screen, so space is scarce.

Yes of course. But you can change the aspect ratio of the elements. It will be much better perceived.

Here is RaSDR http://dl2stg.de/stefan/hiqsdr/rasdr.html  
Unfortunately, my homemade transceiver is not supported by this program.
Here is the design for a 7 inch display for some homemade DDC / DUC transceivers
[image]
[image]

I ask you to excuse me, probably this is offtopic, sorry again.

g4fkk, 2020-06-06:
Out of interest, has anyone been able to get a Pluto working with piHPSDR?  I tried to install the prerequisites this morning following dl1ycf's/John's instructions but got stuck while installing libiio.

Cmake could not find "LibXml2" (although libxml2 seems to exist), and it could not find BISON or FLEX and refused to go any further.  I'm using a Pi4 with Buster (10).

73, Martin - G4FKK

dl1ycf, 2020-06-06:
I run the Adalm pluto with piHPSDR on my Macintosh, I faintly remember that I once tried it on the piHPSDR. But at that time, I remember I had a hard time to manually install and compile all the prerequisites. Note that a PTT *output* on the GPIO would be highly desirable for the Pluto, since any of the hardware solutions proposed result in hot switching unless you have separate RX and TX antennas. If you are really interested, contact my directly dl1ycf@darc.de and I can see if I can get it to run on my Pi-4 with the Pluto.

g4fkk, 2020-06-06:
I've emailed you directly - many thanks and 73, Martin - G4FKK

w0iy, 2020-06-06:
If you get it to compile, please post the results to this list.
Thanks
Barry w0iy

dl1ycf, 2020-06-08:
I have "toasted" a fresh image of Raspian onto a micro-SD card, and the following shell script brings me from this "plain vanilla virgin" system to a pihpsdr running with my Adalm Pluto over the USB connection. However, the sequence of commands is exactly what is in the "Compile from Sources" document (perhap the version in my repo is a little bit more up-to-date, I guess your problem is solved by the commands

sudo apt-get install --yes bison
sudo apt-get install --yes flex
sudo apt-get install --yes libxml2-dev  

But see  the "whole story" in the "install.txt" attachment which is a shell script.

[install.txt](https://github.com/g0orx/pihpsdr/files/4746613/install.txt)

g4fkk, 2020-06-08:
Thanks Christoph,

Yes, those three lines were all that were needed to get my Pluto working with piHPSDR.  The Pluto is flaky on transmit, but that might be my cheap and nasty soundcard.  It's also quite a long way off nominal frequency but not a huge problem at this stage.

Thank you very much for your help.

73, Martin - G4FKK

dl1ycf, 2020-06-08:
The frequency issue is probably not pihpsdr’s fault.
I have listened to a repeater at 439.1125 Mhz that
appears at 439.116 (and varying somewhat).

For the transmit, I will see to it whether it is the sound
card or missing „horsepower“. This I can simulate by feeding
a clean synthetic sine generated inside pihpsdr for debug
purposes.

g4fkk, 2020-06-08:
The frequency issue is definitely not pihpsdr's fault.  I get the same offset using the Langstone transceiver software.  My Pluto has the original not-very-good tcxo.  Eventually I intend to GPS lock it.

It'll be interesting to see what results you get from your testing.

Thanks again for your interest,

73, Martin - G4FKK

> The frequency issue is probably not pihpsdr’s fault. I have listened to a repeater at 439.1125 Mhz that appears at 439.116 (and varying somewhat). For the transmit, I will see to it whether it is [...]

g0orx, 2020-06-09:
I would definitely either use an external reference like the Leo Bodnar GPS Reference Clock or  change the TCXO to a better one.  I am using the Leo Bodnar at 25MHz to feed both the PlutoSDR and the LNB for QO-100.

-- John

ua9yk, 2021-05-24:
pihpsdr installed on ubuntu 20.10 how to start HiqSDR?

dl1ycf, 2021-05-24:
pihpsdr supports HPSDR (version 1 and version 2 protocol),
and SoapySDR devices provided you have installed SoapySDR
support libs for your device.

As far as I know hiqsdr uses a separate protocol. It it
implemented e.g. in quisk.

ua9yk, 2021-05-24:
https://www.youtube.com/watch?v=yj9GEuLxdhI&t=9s

py2rn, 2024-02-14:
Hi! I need some help, trying to run PiHPSDR with my adalm-pluto but got stuck on this screen, I have no external controller, pluto is alive on the network but it seems it does not recognize/connect, any missing step here? (Pic attached)
[image]

Thanks
ed - py2rn

dl1ycf, 2024-02-15:
You have to compile with SOAPYSDR enabled.

Your picture states that you enabled MIDI, SATURN, GPIO
but not SOAPYSDR.

py2rn, 2024-02-16:
I got confused reading some early posts here and supposed that "#" was already removed by default in the initial installation for SOAPYSDR.
After removing the # I also forgot to do "make clean" before the "make" (Thanks to Jac CU2ED)
It is working now with pluto, just need to get some time and finish the software configuration
Thank you
Ed PY2RN

dl1ycf, 2024-02-18:
> It is working now with pluto, just need to get some time and finish the software configuration 
> Thank you 
> Ed PY2RN

So we can close this now. Just a final remark: as long as your Linux distro contains Soapy
modules in version 0.7.x, you have to compile all Soapy libs and radio modules
from scratch (my installation
script does this). If you distro has advanced to Soapy v0.8.x, you can just install Soapy
(and all required radio modules) from the distro.