g0orx/pihpsdr #80: piHPSDR Latest version not saving settings between sessions

mgarrett1955, 2019-11-14:
Hello, thank you for the awesome system!! 

My setting are not being saved between sessions. I have searched around and looked at the manuals etc. but can not figure it out.

g0orx, 2019-11-14:
I suspect you did not delete an old pihpsdr directory. Earlier versions ran the app as root so the props file was owned by root. The latest code does not run as root so cannot overwrite the props file. Deleting old props files should fix the problem.

-- John

PeWa1, 2022-02-23:
My settings is not stored either, where are the props file stored ? I cant find it
And strange is both my IFC Lo and Cable shows MAC 00 01 02 03 04 05  in pihpsdr and the actual MAC when i check with ifconfig is dc a8 52 31 2c b2  
I have VU2DLE´s version for Radioberry console  

//Peter