g0orx/pihpsdr #81: Pursignal problem

chrk1011, 2019-11-15:
The pihpsdr is excellent program for newbee like me..
I learnt so many lessons by following the code.. We are using Apache labs Orion Mk-2 card with old protocol.. We took a branch of pihpsdr code way in the year 2016-17 and included other hardware variants to include the interfacing with our customized PA and HF modem hardware. The program is working fine. Recently we started with Puresignal interface by following recent pihpsdr release. We did all the necessary modifications in our program as per this new release. The algorithm is started working the screen shows "feedbk" level to green but the "correcting" is still black colour only. The status information of puresignal shows the following: cor.cnt is continuosly increasing, the STATUS is changing from RESET, COLLECT, MOXDELAY etc.. the sln.chk is reaching a value of 20 and stays there only. We used the default parameters of the program. 
We need support to overcome this problem

g0orx, 2019-11-15:
I will need some information from you about the hardware. Can you let me have the details of the hardware you added, preferably the schematics.  How have you connected up the feedback from the PA to the Orion Mk-2 card?

Regards,  -- John g0orx

chrk1011, 2019-11-15:
We have a power amplifier capable of 1kw power.. The output of PA is connected through a 30dB directional coupler to 50 Ohm dummy load.. The coupled output is connected to Rx1 input of Orion mk2 card.. I plotted the spectrum output of coupled feedback and Tx feedback separately.. The spectrums are perfect. The Tx feedback is clean without any IMD components. The Rx feedback is having IMD components same as that were shown in spectrum analyzer.

chrk1011, 2019-11-15:
Sir,
Small correction the 30dB coupled output is connected through Rx1 of Orion mk2 card through a variable step attenuate..  We adjusted the step attenuate level such that the feedback level is turning to green colour..
The spectrums of Tx feedback and Rx feedback are plotted in the panadaptor separately.. These plots are inline with the anticipated results.. Rx feedback shows IMD components and Tx feedback is clean..

g0orx, 2019-11-15:
It is really hard to know what is happening.  Can you give me access to your code so I can look at it?

chrk1011, 2019-11-18:
Sir,
There are some difficulties to share the code from my end..
However following observations may give some more information on this matter..
We are checking puresignal with twotonegen command and monitoring the spectral purity in the external spectrum analyzer. Sometimes there are sudden uneven spectral components are appearing i.e. spikes in the external spectrum as if the correction algorithm is being adopted ..

With voice signal, twotonegen off codition, very frequently the correcting indication is turning to green colour but the spectral output in the external spectrum analyzer is not at all good. Additional spectral components are being added..

We need some information  regarding the settings like optimization of AMPDELAY, LOOP DELAY, MOXDELAY, PSPTOL and some other parameters like INTS and SPI etc.. How these parameters affect the PURESIGNAL algorithm.. 
Thank you sir ..

g0orx, 2019-11-19:
With the default setting PureSignal should be working perfectly. It does using the current piHPSDR code with both Portocol1 and Protocol2.  On protocol 1 it does work best at 192000 sample rates,

I don't understand why there is a problem letting me see h source code. Can you not just tar up the source and either email it to me or put it on a a shared resource on the cloud.  You can also create your own github and send  me the link.

I am also interested who you are and what your hardware is?

-- John

dl1ycf, 2019-11-21:
If the "cor.cnt" is constantly increasing, then the WDSP library actually DOES apply corrections. In this case, if you apply a two-tone signal, you should see some "satellites" produced in WDSP whan you look at the outgoing TX signal.

dl1ycf, 2019-11-22:
Oh, before I forget: I have an Anan-7000 here in my lab (Orion-MkII) running old protocol, so I can also do a test if you want.

DL1YCF.

chrk1011, 2019-11-22:
The outgoing signal with two tone test has sometimes spikes in the spectrum.
But it could not able to correct IMD produced in PA with two tone test..
Please let us know with old protocol Orion mk2 will work for not..

dl1ycf, 2019-11-22:
I will put things together and test, but I have tested it with
the Orion-MkII board and it worked flawlessly.

What you should see in the outgoing signal is two satellites
more or less with constant amplitude which are exactly where
the IM3 satellites are.

Give me some time, and I run the test and send you lots of screen
shots, both of the outgoing signal and of the feedback signal.

The feedback signal gives you a good impression how the signal
looks like „at the antenna“, but you cannot take it quantitatively.
But qualitatively it is consistent with what you measure with
a SpecAnalyzer (I have done this).

Please visit my site http://dl1ycf.darc.de,
there you can find lots of experiments with Anan with and without
external PA and with PURESIGNAL/piHPSDR.

However, for the ANAN this was the „new protocol“, but meanwhile I
have switched back to P1 there. But see also the report for
the RedPitaya based SDR running P1, where PURESIGNAL also works
quite well.

g0orx, 2019-11-22:
I have also tested the code running protocol 1 and protocol 2 on my ANAN-8000DLE which uses the Orion Mk II card.  It also works well on my ANAN-100D using the Angelia card.

chrk1011, 2019-11-22:
We are using external power amplifier with 1kW output with an input power of 18dBm from Orion mk2 card. The output power of amplifier is configurable with output power from 25 watt to 1kW with 18dBm input..
We tested this amplifier at 25watt and 100 watt outputs only.. The IMD of PA is around 45dB with 25watt.. The IMD of PA is 40dB with 100watt. The IMD components are being observed with the receiver feedback in panadaptor same as like external spectrum analyzer.. 
We may be doing some mistake with setup or settings but could not able to locate..

dl1ycf, 2019-11-22:
Since the "cor.cnt" is running: if you run the two-tone test and look at either the outgoing signal or the feedback signal, does it change if you hit the "Off" button, then PS should stop working.

And more so, if you disconnect the feedback line at the PA, does the feedback signal go down?
This is no kidding, sometimes there are "magical" feedback paths. Clearly, an unintended feedback of the "clean" input signal will surely lead to no correction.

I am completely stuck here (so is John), since it works with all the hardware we have access to
(but I promis to repeat this with my Anan). But this weekend I opt for going CQ WW contesting .....

dl1ycf, 2019-11-22:
OK, I just did the experiment. Compiles piHPSDR from the sources on my Mac, hooked up ANAN-7000 and dummy load with attenuator, connected attenuator output to "Bypass Input" at the anan.

In the PS menu, check "Auto Attenuate" for good feedback level.

With PS *NOT* enabled, went to the PS menu and hit "TwoTone": I see a clean out-going two-tone signel. Hitting MON button has no effect.

With PS enabled, clicking "TwoTone": after a short while, panadapter stabilizes and I see lots of "satellite spikes" in the outgoing TX signal, after clicking "Off" the TX panadapter again shows the clean out-going twotone signal.

Now click "MON", and the panadpater shows the TX feedback signal, it has some IMD3 and IMD5 satellites from the PA, since PURESIGNAL is still switched OFF. But upon clicking "Restart", the feedback signal after a short while stabilizes and the satellites are gone. Now clicking MON again you see the outgoing TX signal with many spikes.

I probably cannot deposit big image files here, but upon e-mail request (dl1ycf@darc.de) I can send screen-shots from what I just described. And, what I just described is the expected behaviour, the "going away" of the satellites in the feeback signal is also visible on a SpecAnalyzer.

Yours, 
Christoph DL1YCF.

P.S.: If you have a "naked" Orion Board, where do you connect the feedback cable? The way PS is set up is that is assumes the feedback signal goes to the ADC0, because this is the standard setup. If you connect the feedback signal directly to the second receiver (ADC1) then you have to insert a line after line 1506 in old_protocol.c:

1499 #ifdef PURESIGNAL
1500         // if n_adc == 1, there is only a single ADC, so we can leave everything
1501         // set to zero
1502         if (n_adc  > 1) {
1503             // Angelia, Orion, Orion2 have two ADCs, so we use the ADC settings from the menu
1504             output_buffer[C1]|=receiver[0]->adc;                        // RX1 bound to ADC of first receiver
1505             output_buffer[C1]|=(receiver[1]->adc<<2);                   // RX2 actually unsused with PURESIGNAL
1506             output_buffer[C1]|=receiver[1]->adc<<4;                     // RX3 bound to ADC of second receiver
1507                                                                         // RX4 is PS_RX_Feedback and bound to ADC0
1508                                                                         // RX5 is hard-wired to the TX DAC and needs no ADC setting.

dl1ycf, 2019-11-22:
I can add that for me "sln.chk" stays at zero.