g0orx/pihpsdr #86: I2C Compilation Error

OK2MTW, 2019-12-14:
Hello friends,
I have a problem with compilation, see bellow:

Peace of terminal report
/usr/bin/ld: i2c.o: in function `read_word_data':
/home/pi/git/pihpsdr/i2c.c:106: undefined reference to `i2c_smbus_read_word_data'

I have installed all libraries, but I have founded, that i2c-dev.h doesn't include declaration of i2c_smbus_read_word_data or i2c_smbus_write_word_data.

Could you give me any advice, what can I do? Uncle google doesn't help me yet.

Thanks Martin OK2MTW

g0orx, 2019-12-14:
They have removed the inline defines and moved the headers to /usr/include/i2c/smbus.h and functions are in libi2c.so.
I have pushed a change to the Makefile and i2c.c that should fix this.

closed 2019-12-14 by g0orx

OK2MTW, 2019-12-14:
Thanks for all, compilation pass.
73 Martin OK2MTW