g0orx/pihpsdr #91: full-rx_buffer: id=0 fexchange0: errot=-2

db7bn, 2020-01-21:
Dear John,
Dear all,
I'm building an SDR Receiver based on Raspberry PI 3 and RTL-SDR v2 by using the SOAPY SDR lib.

After modifying the Makefile (# uncomment the line below for SoapySDR) and completing the make process I could start the pihpsdr software and select the SOAPY device (rtlsdr (Protocol SOAPY_SDR) on USB). After that the  pihpsdr window starts with the lower part including the sliders. But the upper part of the window is black. By clicking on the menue button another empty black window comes up.

In parallel the terminal screen shows several similar error messages with the following text:
full-rx_buffer: id=0 fexchange0: errot=-2

My software setup:
Raspbian GNU/Linux 10 (buster)
SOAPY libsoapysdr-dev (0.6.1-4)
pihpsdr version 2.07 Beta

Would did I made wrong? Any tip or hint would be helpfull.

many thanks in advance and best regards
Karl-Heinz
-DB7BN-
kh@db7bn.de

g0orx, 2020-01-21:
I would update to the latest SoapySDR.  The version of the shared library that is in the release/pihpsdr directory is libSoapySDR.so.0.8.0. I would also consider updating to an RPi4.

-- John

db7bn, 2020-01-22:
Hi John,
many thanks for the quick answer.

What is the easiest way to update the libSoapySDR.so.0.6.0. into a libSoapySDR.so.0.8.0. version by keeping the rest of the installation untouched?

best regards
Karl-Heinz

db7bn, 2021-10-03:
fixed by using a RasPi4 / 4GB

closed 2021-10-03 by db7bn