g0orx/pihpsdr #93: PureSignal on Hermes Lite 2

rack201, 2020-02-06:
Hi there

PureSignal does not appear to work with the Hermes Lite 2.  Looks like pihpsdr 2 rc2 is not seeing feedback or not at the right level.

Auto-Attenuate does not seem to adjust attenuation and the correcting indicator stays red, regardless of drive level or attenuation setting.  I am using the ’internal’ feedback option.

When running the two tone test, the console fills with ‘Penelope Software version: ##’ if that’s relevant.

Same setup for PS works in OpenHPSDR PowerSDR, but need to change the SetPk value to 0.23 for the Hermes Lite 2 - have tried this in Pihpsdr with no success.

Regards Chris - AG5RR

ra1nb0w, 2021-03-30:
Read https://raw.githubusercontent.com/wiki/softerhardware/Hermes-Lite2/docs/dl1ycf_hl2.pdf
And use the dl1ycf code.

rack201, 2021-03-30:
My bug post was over a year ago.  Christoph provided the revised code in response. All is well.

ra1nb0w, 2021-03-30:
Therefore, you should close this issue.

closed 2021-03-30 by rack201

rack201, 2021-03-30:
I was unaware it was ever fixed in the main code, so didn’t close it. It’s closed now.