g0orx/pihpsdr #94: PiHPSDR Controller MK2

dg5cst, 2020-02-12:
Hello PiHPSDR fans,
just playing with raspberry 4 and Hermes-Lite 2.
Works nicely...many thanks for that great work John and Christoph.
I have seen in the Amsat forum, that John has build a new MKII version of the controller.
He mentioned the the knobs and buttons are connected via a controller board now.
I wonder if there are any infos around? 
I really would like o build a second controller.

Sorry for bandwidth 
73 Seb

g0orx, 2020-02-13:
I will upload the latest schematics when I get back home next week.

g0orx, 2020-02-16:
I have uploaded the schematic and the panel layout for controller 2 in the new directory controller.
-- John

UN7RX, 2020-03-21:
I haven't visited github for a long time , and this is fantastic! Full- fledged controller! Thank you very much for the scheme, John! It turns out just a great transceiver!

UN7RX, 2020-04-15:
Good afternoon! Please tell me, if I have 1 receiver, can I assign other functions to encoders?

g0orx, 2020-04-15:
All encoders and buttons can be configured for different functions.

UN7RX, 2020-04-16:
Wonderful, thank you very much!

VK3JKY, 2020-09-04:
Hi there I am a true newby! Joined GitHub today.

Is there some document that gives me some idea on what new features the Mk 2 version offers? I was about to build a M<k 1 version with a 10.1" display but I have put that on hold for the moment.

VK3JKY, 2020-09-05:
> Take a look at the schematics in the the controller directory (
> https://github.com/g0orx/pihpsdr/blob/master/controller/PiHPSDR_Controller_MK_II_Rev.2_Schema.zip).
> The difference is that it has more encoders (4 dual encoders plus the VFO
> encoder) and more buttons using an MCP23017 on the i2c bus. The current
> software works with any of the versions of hardware including one with no
> encoders or buttons by a selection on the startup screen. The Mk 2 encoder
> and button functions can be configured by the user in the software.

Thanks John,

I will probably have some further queries down the track.  Impressed with
your concepts in the new controller design.

I have been programming for industrial projects using Delphi for the last
20 years but apart from some dabbles in Arduino type C I am very much of a
neophyte regarding the C language. What C compiler do you use or recommend?

Thanks again.
Regards

g0orx, 2020-09-05:
On the Raspberry Pi I use the standard gcc compiler/linker that is part of
the OS distribution.

-- John