g0orx/pihpsdr #114: Problem with pihpsdr

UN7RX, 2020-04-18:
Hello.
I had version 1.2.3.
I updated pihpsdr to version v2.0.0-rc7.
The program has stopped starting. It does not start at all, when the pihpsd button is pressed, nothing happens on the desktop.

I deleted the home / pi / pihpsdr folder. Created pihpsdr folder again, installed version 1.2.3.
Now at startup, the version check window opens and the Start button appears. When I press the Start button pihpsdr hangs completely. The processor is heavily loaded. This continues until I forcefully kill the program through the dispatcher. What could be wrong? Raspberry operating system updated to the latest version.

sq8mvy, 2020-04-23:
Hi,
Please, run pihpsdr in the console and show logs.
Pihpsdr v2.x.x works for me correctly on RPI4, Odroid N2, Kubuntu and Linux Mint 19.2

g0orx, 2020-04-23:
You may also want to try removing any existing props files As there have been several changes in the latest test version.

UN7RX, 2020-04-23:
Hi!
File in home / pi / pihpsdr / log folder?
At first there was information in the log that there were insufficient permissions to run the usr / bin / pihpsdr file. I changed the permissions to 755.
Now the following information is in the log:
/ usr / local / bin / pihpsdr: error while loading shared libraries: libi2c.so.0: cannot open shared object file: No such file or directory

sq8mvy, 2020-04-23:
hmm, weird...
Check if it changes anything:
sudo ldconfig -v

g0orx, 2020-04-23:
It sounds like you are running an old version of Raspbian.  I would suggest 3 things:
1. Update to latest Raspbian,
2. Delete or rename the existing pihpsdr directory.
3. Extract the tar file to create a new pihpsdr directory and run the install.sh script.

Then try running pihpsdr again.

UN7RX, 2020-04-24:
Good afternoon!
I had the latest version of Raspbian Stretch. I deleted it and installed Raspbian Buster. I re-installed pihpsdr 2.0-rc7 and it immediately started!
Thank you so much for your help!
One more question, is it normal that the program freezes if I choose the NR2 noise canceller?

sq8mvy, 2020-04-24:
Hi,
Raspbian Stretch does not support RPI4 hardware, but will run on it.

> One more question, is it normal that the program freezes if I choose the NR2 noise canceller?
For me NB, NB2 and NR, NR2 do not freeze pihpsdr.

UN7RX, 2020-04-24:
NB, NB2 and NR working good , problem is only NR2...

g0orx, 2020-04-24:
The NR2 problem is fixed in the source code.  I will try to push an updated binary version on soon.

UN7RX, 2022-04-24:
Hello! My program still hangs when I turn on the NR2. Rb PI 4, 4Gb memory. The Raspbian version is the latest, with updates. Nothing but OS and piHPSDR installed.
Someone solved this problem? v2.0.0-rc15

f6dro, 2024-12-17:
Hi , tnx for help , I'll look around agn. I must say that my Rpi os is the latest 64 bits os downloaded yesterday