g0orx/pihpsdr #115: Gerber files - wrong file extensions

ExplodingLemur, 2020-04-25:
The gerbers in https://github.com/g0orx/pihpsdr/blob/master/controller/PiHPSDR_Controller_MK_II_Rev.2_Gerber.zip appear to be wrong.  I've mapped the outline and layers, but I'm not sure which is the correct drill file (mainly because I don't know what a drill file should look like :) )

.g0 -> .gto
.g1 -> .gts
.g2 -> .gtl
.g3 -> .gbl
.g4 -> .gbs
.g5 -> ,gbo
.g6 -> .gko

Remaining unknown files in the archive: .d0, .g50, .ipf, .rep

ExplodingLemur, 2020-05-04:
I think .g50 -> .txt will make it recognizable as a drill file.
I'm not able to get OSHPark, JLCPCB, or PCBWay to recognize the outline though (.g6/.gko)

gerryk, 2020-05-08:
They are not wrong, they seem to be UltiBoard Gerber files. 
It seems that the drill files are not the same as 'standard' Excellon drill files. It seems that the 'ipf' file contains the drill tool paths.
There are board fab houses that will accept UltiBoard files, but they are much more expensive than my usual vendor.
Converting from Ultiboard is not a runner, it seems.

ExplodingLemur, 2020-05-12:
I was able to pull the other layers into KiCAD's GerbView, export to PCBNew, and then re-export the gerbers.  The drill file is still a problem, though.
[gerber.zip](https://github.com/g0orx/pihpsdr/files/4613357/gerber.zip)

gerryk, 2020-05-12:
I tried renaming .ipf to .drl and got a valid drill file, but all drills were 0.5mm, which is wrong.
From what little I have been able to determine, the tool paths and drill diameters are in different files. I haven't tried combining them though.
However, this may all be moot, the thread here discussed this, and proposes some good solutions, including a source of 'regular' style gerbers: https://groups.google.com/d/msg/hermes-lite/i5SNriEviHM/xadUKS3YBQAJ

ExplodingLemur, 2020-05-12:
Trying http://k9ivb.net/PiHPSDR_Project/PiHPSDR_MK_II_Rev.2_JLCPCB_Gerber.zip from http://k9ivb.net/PiHPSDR_Project/index.htm

ExplodingLemur, 2020-05-20:
That archive worked for JLCPCB, but it did take a few tries.

ExplodingLemur, 2020-05-27:
> 
> 
> Trying http://k9ivb.net/PiHPSDR_Project/PiHPSDR_MK_II_Rev.2_JLCPCB_Gerber.zip from http://k9ivb.net/PiHPSDR_Project/index.htm

Workaround successful.

closed 2020-05-27 by ExplodingLemur