g0orx/pihpsdr #117: Application crashes with Arduino CAT Controller for HPSDR

Ujjwaana, 2020-05-16:
1. Have the basic HPSDR CAT controller by N6LYT
https://www.tapr.org/pdf/DCC2015-ArduinoCAT-Controller4HPSDR-G0ORX-1.pdf
https://github.com/TAPR/OpenHPSDR-SVN/tree/master/N6LYT/Arduino/hpsdr_basic_controller
2. Connect the Arduino Nano with the Rotary controller to Raspberry Pi through USB. You can see the Arduino emulated as /dev/ttyUSB0 and CAT commands like ZZSA;ZZSA ..... ZZSB;ZZSB appear  on Serial monitor
3. Now launch Pihpsdr and Click Menu->RigCtl->Check CAT over Serial -> Select ttyUSB0 for device and baud rate as 9600 
4. Now move the Rotary encoder. After few rotation , Pihpsdr will crash 

I am attaching the Console log with all Debug flags in Pihpsdr enable during compile and the core dump (.BZ2 Zipped for the sake of Github letting the file to upload) 

[pihpsdr_dbglog.txt](https://github.com/g0orx/pihpsdr/files/4639268/pihpsdr_dbglog.txt)

[core.zip](https://github.com/g0orx/pihpsdr/files/4639269/core.zip)

g0orx, 2020-05-20:
I have pushed an update (v2.0.0-rc10) with a rewrite of the rigctl (CAT) code.  It is still a work in progress, but a major number of the CAT commands are implemented.  More testing is in progress but the basic code is working OK.  Let me know if there are any problems.

df5sf, 2020-05-21:
Hello,
tested it with fldigi (4.1.12):
Frequency change is ok in both directions.
changing the mode from fldigi has some problems:
e.g. USB is ok, LSB is ok, CW changes to DSB, CWR changes to AM, FM changes to CWL 

tested with WSJT-X (V2.1.2):
if I change the band sometimes piHPSDR freezes and WSJT-X stops with 'Rig Control Error'. Not every time.
In debug mode the last commands see attacched file.

73
Uli
DF5SF

[image]

dl1ycf, 2020-08-14:
This should be resolved now. It is a hamlib issue (too small timeout). In Hamlib, this is now increased considerably.