g0orx/pihpsdr #118: RIGCTL faulty ??

df5sf, 2020-05-17:
Hello
I have piHPSDR V2.0.0-rc9 and rigctl from Hamlib 3.3 on a raspberry Pi 4. Tested rigctl with the following command:
rigctl -m 240 -r localhost:19090 <command>

command f get_frequency is ok
command J +100 set RIT is ok
....
I think the interface is working ok.
But command F set frequency is not working, nothing happens
rigctl -m 240 -r localhost:19090 F 00014200999 
For my application this is the most important command.

Any ideas ?

73
Uli
DF5SF

g0orx, 2020-05-17:
See issue #117. I am in the process of rewriting the interface as there were too many problems and omissions with the current code.
-- John

closed 2020-05-17 by g0orx

g0orx, 2020-05-20:
I have pushed an update (v2.0.0-rc10) with a rewrite of the rigctl (CAT) code. It is still a work in progress, but a major number of the CAT commands are implemented. More testing is in progress but the basic code is working OK. Let me know if there are any problems.