g0orx/pihpsdr #123: receiver pre-amp

human-blip, 2020-07-26:
Is it possible to make the receiver pre-amp switchable?

g0orx, 2020-07-26:
The preamp can be switched on/off in the Menu->RX

[image]

human-blip, 2020-07-27:
I'm not seeing the Preamp checkbox for either of my radios.

First screenshot is for ANAN100 running protocol 1.

[image]

Second screenshot is for ANAN100D running protocol 2.

[image]

g0orx, 2020-07-27:
For some reason I though you were using an ATLAS based system.   For a Hermes/Angelia card the preamp on/off is disabled and the attenuator is used instead.  Setting the attenuator slider to -20dB is the equivalent to turning the preamp off.

human-blip, 2020-07-27:
Applying attenuation makes an audible  difference but does not change levels on the S-meter or the panadapter or waterfall.

dl1ycf, 2020-08-14:
The receiver preamp is not switchable in all these radios except the first-generation boards. The hardware simply is not there. Therefore you do not get the choice in the menus, doesn't this make sense? -- 73, dl1ycf.

dl1ycf, 2021-01-26:
BTW, using a preamp or an attenuator does not change the real signal out there in the air, so this SHOULD NOT change S-meter and panadapter readings. A -73 dBm signal from the antenna jack should display as -73 dBm no matter what
you do with preamps and attenuators (some programming effort went into this!).