g0orx/pihpsdr #131: Tinkerboard v2.0.0-r15 no sound

dm6tt, 2020-10-18:
I installed the latest version on a fresh Tinker_Board-Debian-Stretch-V2.1.16-20200813 image.
PiHPSDR was compiled on the tinkerboard, since I the binary thoroug a error due to an old glibc.

Everything is running fine, but the sound does not work.
I installed a USB CM108 sound module that works fine directly out of X11.

The log show no error:

local_output_changed rx=0 plughw:2,0 USB PnP Sound Device
local_output_changed rx=0 local_audio=0

[image]

linaro@tinkerboard:~$ lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0bda:481a Realtek Semiconductor Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

inaro@tinkerboard:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: OnBoard [USB Audio OnBoard], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: OnBoard [USB Audio OnBoard], device 1: USB Audio [USB Audio #1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: OnBoard [USB Audio OnBoard], device 2: USB Audio [USB Audio #2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: rockchipminiarm [rockchip,miniarm-codec], device 0: ff890000.i2s-i2s-hifi i2s-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Any hint what could be the problem?

Thanks
73, Marcus, DM6TT

g0orx, 2020-10-20:
I just rebuilt a Tinker Board from scratch and I am not having any problems with audio.  Based on your screen dump, did you enable Local Audio by clicking on the checkbox?

[image]

-- John

dm6tt, 2020-10-20:
Hi John,
thanks for the reply.
I have two radios running the Hermes protocol.
With my Red Pitaya running an older version of Pavels TRX I get no local sound  working. This is independent from the 'Local sound' switch. The pitaya has the IP .156
I have a LanIQSDR in my network that can be reached via PiHPSDR as well. Here the sound is working when enabling the 'Local Audio Output switch'. The LanIQSDR has the IP .196
They both use different versions of the protocol.
[image]

I will update the pitaya to the latest version and will give you an update here.

Thanks
73, Marcus

dm6tt, 2020-10-23:
Hi John,
I tested it again using a red-pitaya with the actual image from pavel red-pitaya-alpine-3.12-armv7-2020062.
The sound is not working with this device, but without any problem using the LanIQSDR,  Protocol1 v2.4.

Can anyone confirm that sound is working with a red-pitaya in his environmet?

Thanks
73, Marcus

g0orx, 2020-10-23:
I do not have a red-pitaya so cannot test with it. I do have an ANAN-100D, ANAN-8000DLE, Hermes Lite 2, IquadLabs IQ2, RadioBerry and a bare HPSDR Orion board.  All are working with no problem using piHPSDR on the Tinker Board, so I am at a loss as to why this is not working.  If the IQ samples are being received and displayed correctly then the audio should be heard provided the correct Local Audio device is selected and the AGC and the AF gain settings are set correctly.

Could you send the log file with the Local Audio disabled and then enabled?

-- John

dl1ycf, 2021-11-10:
I use an RedPitaya SDR (Stemlab/Hamlab) with piHPSDR with no problems. If the signals are on the spectrum, then the whole audio chain downstream does not depend on the radio. If your radio has no codec, then of course you have to
engage "local audio output".  It is more likely that you have a problem with your sound cards and/or pulseaudio. Perhaps the pulseaudio daemon is not running?