g0orx/pihpsdr #133: Gerber File

hb9trt, 2020-12-09:
I tried to work with the Gerber file but have troubles to open it. I plan to make some boards for our Ody2 project and PiHPSDR. I tried to oben the Gerber in Sprintlayout and also upload it at a maufacturer. Seems to be a special format (?). Do you have some more informations about the Gerber Format of the controler mk2?

g0orx, 2020-12-09:
Can you contact Kjell la2nikjell@gmail.com who will try to help you as he
was the author of these.

-- John

hb9trt, 2020-12-09:
Hello John, 

Congratulation for the fine SDR Project for the Raspberry PI. Also thank you that you share it with us. 
I’ll try to reach Kjell for this question. 

Reto

barkerb1, 2021-01-04:
There's an issue with the file extensions in the gerber files, K9IVB has. already resolved the issue - https://www.k9ivb.net/PiHPSDR_Project/index.htm

Corrected gerber files are at the bottom of the page:
https://www.k9ivb.net/PiHPSDR_Project/PiHPSDR_MK_II_Rev.2_JLCPCB_Gerber.zip

This issue should be fixed in the source as the average hobbyist will struggle trying to figure this issue out.