g0orx/pihpsdr #134: Screen Resolution

hb9trt, 2020-12-09:
First I tried with a 7 Inch Screen with a resolution 1024x600 and there's no possibility to expand pihpsdr to full screen. Now I tried it with a lower resolution 800x480 where full screen Ddisplay of PiHPSDR works. But with the look and feel of a lower resolution display, some lines or letters are pixelated. Is there a possibility to increase the resolution for PiHPSDR?

g0orx, 2020-12-09:
The display is designed to work with the Raspberry Pi 7 inch touch screen
which is 800x480. The source available for users to modify for other screen
sizes.

-- John

hb9trt, 2020-12-09:
Thanks for the answer. I'm not a good programmer. But I can try ;)

UN7RX, 2021-01-16:
Here is information on how to change the screen resolution for pihpsdr
https://charly25-sdr.github.io/software/pihpsdr

TeiCon67, 2021-08-16:
And here is an Link with an Guide how to Compile on an Raspberry:
[https://github.com/g0orx/pihpsdr/wiki/Compiling-piHPSDR-on-Raspberry-Pi-running-Raspberry-Pi-OS](url)

Charlier26, 2021-10-22:
I'm trying also to get it to fit 1024x600 screen.  Once I changed (in main.c) the MAX_SCREEN definitions to 1024 and 600 it resized. But I changed it back and cannot get it to do it again after recompiling. The suggestions above do not match the current contents of main.c....... Any help?