g0orx/pihpsdr #135: Midi ingores CHAN parameter

DK4FT, 2020-12-21:
I am using a Numark DJ2Go2 Controller, where the Key numbers are not unique 
and the CHAN parameter is needed to identify the right Kay/Knob/Wheel.

The CHAN Parameter is not working in the config of pihpsdr 2.0.0-RC15, as a result only few knobs could be used...

Here an example of the ouput of aseqdump:
- left Wheel
- right Wheel
- left keys 1-4
- right keys 1-4

```
pi@raspberrypi:~ $ sudo aseqdump -p 'DJ2GO2'
Waiting for data. Press Ctrl+C to end.
Source  Event                  Ch  Data
 28:0   Control change          0, controller 6, value 127
 28:0   Control change          0, controller 6, value 127
 28:0   Control change          0, controller 6, value 1
 28:0   Control change          0, controller 6, value 1
 28:0   Control change          1, controller 6, value 127
 28:0   Control change          1, controller 6, value 127
 28:0   Control change          1, controller 6, value 1
 28:0   Note on                 4, note 1, velocity 127
 28:0   Note off                4, note 1, velocity 0
 28:0   Note on                 4, note 2, velocity 127
 28:0   Note off                4, note 2, velocity 0
 28:0   Note on                 4, note 3, velocity 127
 28:0   Note off                4, note 3, velocity 0
 28:0   Note on                 4, note 4, velocity 127
 28:0   Note off                4, note 4, velocity 0
 28:0   Note on                 5, note 1, velocity 127
 28:0   Note off                5, note 1, velocity 0
 28:0   Note on                 5, note 2, velocity 127
 28:0   Note off                5, note 2, velocity 0
 28:0   Note on                 5, note 3, velocity 127
 28:0   Note off                5, note 3, velocity 0
 28:0   Note on                 5, note 4, velocity 127
 28:0   Note off                5, note 4, velocity 0
```

```
I setup the following Config incl. the correct Channel parameter, but each key with the same Key Number triggers the event.
#
DEVICE=DJ2GO2
#
# LeftWheel = VFO "A"
# Channel 1 Contr 6     1 -> turn right / 127 -> turn left
#
CTRL=6 CHAN=1 WHEEL THR=1 -1 -1 -1 64 127 1 63 -1 -1 -1 -1 ACTION=VFOA          # Big wheel:    : main VFO knob
#
# Right Wheel = VFO "B"
# Channel 0 Contr 6      1 -> turn right / 127 -> turn left
#
CTRL=6 CHAN=0 WHEEL THR=1 -1 -1 -1 64 127 1 63 -1 -1 -1 -1 ACTION=VFOB          # VFOB on right big wheel
#
#
#
KEY=27  CHAN=0 ACTION=NONE		# Headphone
#
KEY=1  CHAN=4 ACTION=VFOSTEPUP  	# 1
KEY=2  CHAN=4 ACTION=VFOSTEPDOWN	# 2
KEY=3  CHAN=4 ACTION=NOISEBLANKER		# 3
KEY=4  CHAN=4 ACTION=NOISEREDUCTION	# 4
#
# rightKeys
#
KEY=0  CHAN=1 ACTION=BANDUP 	  	# play
KEY=1  CHAN=1 ACTION=BANDDOWN		# CUE
KEY=2  CHAN=1 ACTION=NONE		# Sync
#
KEY=27  CHAN=1 ACTION=NONE		# Headphone
#
KEY=1  CHAN=5 ACTION=NONE  		# 1
KEY=2  CHAN=5 ACTION=NONE		# 2
KEY=3  CHAN=5 ACTION=NONE		# 3
KEY=4  CHAN=5 ACTION=NONE		# 4
#
```

g0orx, 2020-12-21:
Looking at the documentation and the code from Christoph, DL1YCF, it looks like the channel number specified in the configuration should be between 1 and 16.  The code actually decrements the value defined by 1 so having CHAN=0 becomes -1 which means match any channel.  Could you try specifying the channel number in the configuration as the actual channel number + 1 to see if that works OK.

-- John

DK4FT, 2020-12-21:
Thanks a lot John - this did the trick. :+1: 
Added +1 to each channel value and all controls are working as expected now

Probably worth to mention in the midi documentation :-)

mni tnx, 73 de Alex, DK4FT

closed 2020-12-21 by DK4FT