g0orx/pihpsdr #152: Need ability to program HL2 Clk1 and CLk2 frequencies

K7MDL2, 2021-09-20:
CLk1 can be used for an external frequency refence. 10Mhz is popular choice but others are possible.  Clk2 output can be used to drive another HL2 (default) but can also be used as the LO for a transverter board (with filtering or choosing an integer divisor to remove spurs).

In Quisk you can modify the hardware.py file it load on startup.  ThetisHL2 Beta 5 now has a UI to set both clocks.  For SparkSDR and other SDR app that do not yet know how to set the clock I use a modified hermeslite.py script before loading the app.

This approach can also be used but is clunky.

Propose adding at minimum a configuration parameter to set these clocks.  Best case is a UI to (optionally) set these.  ThetisHL2 Beta 5 is a good example.

I have started looking into the code to find out where the clocks are set and hard code change to test with.

K7MDL2, 2021-09-20:
As a workaround I am running the hermeslite.py script.  It has changed recently and now requires the module netifaces to be installed (pip install netifaces).  

Also on my new RPi Raspian installation Python2 was default and hermeslite.py would fail to find or load some modules.  Ran command below to make Python3 the default, installed netifaces, and now the script works.  Created a desktop app shortcut to the script.

sudo update-alternatives --install /user/bin/python python /usr/bin/python3 10

m5evt, 2021-09-23:
Despite not being a regular piHPSDR user, I offer the following comment on this.

I added HL2 I2C read write to my linHPSDR fork. It is not a quick thing to implement and you really need to study and understand the HL2 protocol wiki page. 

linHPSDR has roots in piHPSDR so I think there is overlap in methods and my code should port across. The code is structured to send cyclic protocol 1 commands. This mean that for the HL2 i2c commands (which are outside the "official" HPSDR protocol 1), you need a "one shot" action to send the command, otherwise you will continuously be enabling the clock. You would need to modify old_protocol.c in piHPSDR.

You can see the commit where I did the basic implementation of i2c read/write [here](https://github.com/m5evt/linhpsdr/commit/117fa731741fc5e1267eb4afaa23aea5382947e1). Perhaps there is a simpler way to do this? (Note, I deliberately avoided opening a port on 1025 because I feel this should be reserved for users and not be touched by PC SDR applications).

The branch with all this functionality in is here:

https://github.com/m5evt/linhpsdr/tree/versaclock

Good luck.

K7MDL2, 2021-09-23:
Thanks for the pointers, I will examine that code. I2C code already exists to do the OC/J16 commands for the likes of a N2ADR filter board.  Have to make sure to send the I2C command to the right I2C bus, number 1 or number 2.

Related to this effort I plan to extend the number of pins via the I2C port expander route to support sequenced PTT and band decoder output in either BCD or parallel output format.  Placing this GPIO expander the HL2 makes the most sense to me sine it may be remotely near transverters, amplifiers, preamps, rotator control, and antenna/switches while the control point is at a desk or table somewhere else.  The 10Mhz ref would also be there, keeping the control point with minimal hardware and still work with a PC SDR app as well.   Hope to start work on this in October, right now I am finalizing the mechanical packaging, searching for a more cost effective solution than milled plate aluminum for a V2 controller PCB I have built and running now.