g0orx/pihpsdr #153: Segfault when selecting 'Step' settings

madpsy, 2021-09-26:
Running latest version on Raspbian. Whenever I click the 'Step' option in menu it instantly hits a segmentation fault. No other output to be seen so not sure how to debug further. 

Where are the settings stored? I tried the usual suspects like ~/.config to no avail. I'm thinking maybe try deleting them and see if it still occurs.

Thanks

g0orx, 2021-09-27:
Fixed in latest push.  Radio properties are stored in ~/pihpsdr with a file name of the MAC_ADDRESS.props (i.e. something like 00-1C-C0-A2-13-DD.props).

closed 2021-09-27 by g0orx

madpsy, 2021-09-27:
Thanks for info. Was that pushed to master? I've just recompiled and still get a segfault even after removing the props file.

g0orx, 2021-09-27:
Yes, it was to the master branch. Did you do a git pull to get the update?
--John

g0orx, 2021-09-27:
I just checked and it does not appear to be on github. I am in London all
day but will check when I get home tonight.

-- John

madpsy, 2021-09-27:
Thaks John - it was definately latest I was using as was from a fresh clone on a different Pi.

db7bn, 2021-09-27:
Hi John,
I belief you updated the GPIOD branch. In the GPIOD branch the file step_menu.c was updated 5 hours ago.
To be honest, I did not understand the difference between the MASTER and the GPIOD branch. Maybe you can explain this in the README.md
best regards
Karl-Heinz
DB7BN

madpsy, 2021-09-29:
@db7bn Nice spot. Can confirm this commit fixes it:

https://github.com/g0orx/pihpsdr/commit/35e50e1c36849a0cd299837b36c9df2e223dba27#diff-a0160de3dd5c4bf6fe95e2c031e523ee0d897fd194f0d9e58645c024c9b06aa4

As you said, looks like John just pushed to the GPIOD branch.