g0orx/pihpsdr #155: Intended operation of encoder switches?

K7MDL2, 2021-09-30:
I found on my unit using a couple standard cheap single encoders with switches, I think the debounce is not used, or not enough.  Most of the time they operate as a momentary push switch.  

For example I assigned MOX to one of the encoder switches and I can 9 out of 10 time push to talk, let go to RX, so it is non-latching. The touchscreen MOX button is latching.  However, once in a while, the switch does latch.   If I assign the Band Menu or Filter Menu function to an encoder switch, it tends to latch 7 out of 10 times.  Different behavior, but still not consistent making me think it is debounce setting related.   I have the GPIO set for pull up, not using any external pullups. 

This causes me to wonder what the intent was, are these supposed to be momentary switches, or latching switches?  If latching then need to look at the debounce treatment and if it is applied with the switch or the function assigned.

One way or the other, neither is working correctly so creating this issue to clarify and track resolving it.  I will poke around the code as well.

K7MDL2, 2021-09-30:
Investigating this more, debounce is used in gpio.c.  the var settle_time is set to 50ms.  I found making this 150ms I got consistent behavior and it appears, by observation, the encoder switch is momentary, not latching.  Not sure what the intent was.  

Press the switch and hold, the assigned Band Menu is shown.  Release the switch and the Band Menu window is removed.  Same results for RIT On/Off and Mute.

This is using a non-PCB Controller1 configuration with 3 push encoders and no standalone switches. I do not have push encoders on my PCB version to test with.

K7MDL2, 2021-10-01:
I believe I solved the problem. The (regular) switches record millis into a switch.debounce value for each switch. The encoder switches have the same structure for switch state but the code to use it was missing in gpio.c

Further, at 100ms settle_time value, the encoder switch acts like a momentary switch. At 200ms, they now act like a latching switch.  Not sure why yet but I will take it for now.

About line 549 in gpio.c is the push switch check.
Below is the change

    } else if(encoders[i].switch_enabled && encoders[i].switch_address==offset) {
      g_print("%s: found %d encoder %d switch\n",__FUNCTION__,offset,i);
  
//new code added to debounce push switch 
    t=millis();
      found=TRUE;
      if(t<encoders[i].switch_debounce) {
          return;
      }
      encoders[i].switch_debounce=t+settle_time;
// end new code

      PROCESS_ACTION *a=g_new(PROCESS_ACTION,1);
      a->action=encoders[i].switch_function;
      a->mode=value?PRESSED:RELEASED;
      g_idle_add(process_action,a);
      found=TRUE;
      break;
    }
.  

Additionally at line 147 the debounce settling time is set

long settle_time=200;  // ms

plmathews, 2021-11-17:
I have incorporated your new code and my encoder switches now work fine - thank you for sharing your code.

K7MDL2, 2021-11-17:
The solution has been working well for normal switches that you release for action.

For gpio PTT (only) where you need action and debounce on both edges, level sensing polling code is needed.  For gpio PTT I found the best results are to set settle time to 0ms and use hardware debouncing.  An exception needs to be added to the debounce code to ignore gpio PTT.

K7MDL2, 2021-11-28:
I have now implemented  MIDI keyer config via the CW menu and will have the Teensy MIDI keyer handle PTT (with debounce) and CW.  Also takes care of lack of gpio pins for Controller 2 V2 PCBs if you have dual shaft encoders with push switches.  I am using a Teensy4.0 with PJRC sound card so it is small enough to stick inside the controller. Mic PTT, Mic Audio and headphone/speaker amp also connected. Power is from USB. 
So the PTT exception is bypassed in this solution and several benefits gained.