g0orx/pihpsdr #161: VFO Encoder Divisor not working

plmathews, 2021-11-16:
I've just installed the latest version of pihpsdr on my Pi/14 bit pitaya system. All went fine but I find that the vfo encoder divisor is not working and it is needed with the 600ppr encoder I am using. Am I doing something wrong please?  G4WJH

plmathews, 2021-11-22:
Perhaps I should add......

I created a new SD card with the Raspberry Pi O/S and then followed the instructions in the Wiki to install pihpsdr Build date 2021-10-28 Build version v2.0.8-rc1 and WDSP version 1.18. The installation went absolutely according to the instructions. The hardware is a Raspberry Pi 4 (8 GB RAM) and a Stemlab 14 bit Red Pitaya. The PI has a 600ppr VFO encoder connected as well as 3 low resolution encoders with switches. All of these encoders and switches work. The only problem is with the VFO Encoder Divisor. I can set the divisor in Menu>Radio to anything from 1 to 60 but this has no effect on the "sensitivity" of the encoder. It feels as though the divisor is stuck on 1 which makes the VFO encoder much too coarse. 

Could anyone help to resolve this problem please.

g0orx, 2021-11-22:
I have pushed a fix to handle the encoder divisor.

plmathews, 2021-11-22:
Many thanks for the fix John. I think there is still a problem though. The Divisor now works in that it adjusts the coarseness of the VFO encoder but when you adjust the VFO frequency (say upwards) by turning the encoder knob the frequency keeps changing upwards when you have stopped turning the knob. It only stops when you turn the knob to reduce the frequency and it then continuously reduces the frequency even though you are no longer touching the knob.  Does this make sense?

plmathews, 2021-11-22:
PS    The effect described in my previous post is not apparent when the Divisor in Menu>Radio is set to 1. It is present at all other Divisor values.

closed 2021-11-24 by g0orx

plmathews, 2021-11-24:
> Latest push should fix this. Test OK on my Controller 2.
> -- John
>
>
>
>
>  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> Image removed by sender.
>
> Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

That's fixed it John. Very many thanks for the fixes and for all of your brilliant pihpsdr code.

Paul