g0orx/pihpsdr #166: Cannot connect audio between PiHPSDR and wsjtx for ft8

morrisford, 2021-11-28:
Trying to get ft8 running. I have done this with old version of pihpsdr and wsjtx 2.2 and earlier. I want to pick a route in pihpsdr for rx and tx and do the same in wsjtx but I cannot get it done. pihpsdr shows only built in audio ports and wsjtx doesn't work when trying to select ports. I am currently using wsjtx 2.5.2

morrisford, 2021-12-04:
I have not been able to resolve this. pihpsdr never sees cards added with "modprobe snd-aloop". I have been looking down in the c code mainly in and around pulseaudio.c and what I see is that modprobe additions just aren't there. This is my first real look at pulse/alsa c code so I could really use some help.
Thanks
Morris Ford K7LSV

dl1ycf, 2021-12-29:
Well, pihpsdr now uses pulseaudio, so you have to use the "monitor" devices not the ALSA loopbacks.
Look in github dl1ycf pihpsdr, there you find detailed instructions in release/documentation.