g0orx/pihpsdr #180: Sudden jump of the noise floor to the top of the spectrum display

kihwal, 2022-12-15:
This happens occasionally. When this happens the actual audio output is affected too. E.g. wsjt-x would show a constant high audio input level.  Restarting the app or changing the sampling rate from the RX menu restores the receiver function.

I am not sure whether my ANAN-10 (non-E version, running protocol 2) is at fault or the software.  It will be of great help if someone can suggest what to look for in pihpsdr.  I can add debug statements and inspect some variables.

cosmo1stgen, 2023-01-14:
I am also having this issue.  It has happened twice both when the sample rates were 384 or greater.  Last time it happened with both receivers open I was transmitting on rec1 when returning to receive rec2 noise floor jumped to 10 over closing rec2 and reopening did not resolve it.  Rec1 was unchanged.  I am not running a pi but a pc on latest linux mint.

cosmo1stgen, 2023-01-14:
Also running hermes 16 bit on protocol 2.

dl1ycf, 2023-09-14:
I had this problem quite frequently with my Anan-7000 running P2. This is a firmware timing problem. It has been
much improved in the latest Orion2 firmware. In the dl1ycf version there is a "RESTART" button which restarts
the protocol, and this was introduced exactly to recover from such situations.