g0orx/pihpsdr #182: ibusbio: cannot find ozy device

murray-lang, 2023-03-19:
My USB OZY is not being found.
~/pihpsdr/log says:
    `libusbio: cannot find ozy device`
lsusb says:
    `Bus 001 Device 010: ID fffe:0007 Free Software Folks HPSDR OZY Rev 5`

What am I missing please?
Cheers,
Murray VK6HL

murray-lang, 2023-03-19:
It turns out the problem is with `libusb_open_device_with_vid_pid()`.

The documentation does say, "_This function has limitations and is hence not intended for use in real applications_".

I found that by using `libusb_get_device_list()`, I could successfully discover the OZY interface.

If I was to make a wild guess, I'd say that issue is that I'm running a 64 bit bullseye, and the libusb code in `libusb_open_device_with_vid_pid()` is making a non-portable number comparison with the IDs.

Anyway, now I'm getting a `LIBUSB_ERROR_ACCESS` error opening the interface, which I hope to fix in the rules file.
Fingers crossed.

dl1ycf, 2023-09-14:
Have you solved the problem? I worked on the OZY interface some time ago in my fork, and had contact with
a guy with OZY hardware and it worked there, but if you also have access to such hardware I would like to test things.

murray-lang, 2023-09-15:
> Have you solved the problem? I worked on the OZY interface some time ago in my fork, and had contact with a guy with OZY hardware and it worked there, but if you also have access to such hardware I would like to test things.

I eventually managed to talk to Ozy OK. It was a while ago, and I can't remember the details, but I think the remaining issues were more around the rules file etc.

However, I never managed to get HPSDR running with PiHPSDR. The code doesn't do the handshaking correctly, so I couldn't get a proper IQ stream coming through. It seemed to me at the time that most of the work on PiHPSDR was around Hermes, and no regression testing had been done to see if Ozy still worked, which it doesn't. That's just my experience and my guess.

I'm certain that it's possible to solve this, but I have limited time and energy after work, and I got exhausted trying to make PiHPSDR talk nicely to Ozy/Mercury. The protocol seems a little flaky, requiring you to send the same commands multiple times to get through. In the end I decided to focus on using my other SDR devices (SoftRocks, FunCube etc.) to play with SDR software.

At some point I'll return to Ozy/HPSDR on the Pi. When I have a lot more free time.

Good luck

dl1ycf, 2023-09-15:
Please first try my fork. The problem is I do not have access to such legacy
hardware. I had contact with a ham with a "magister" board and die some
software changes in January 2022 so these changes are not in John's repo.

I got an email with the following screen-shot

there were still problems, but perhaps you start testing my version.

murray-lang, 2023-09-15:
Thank you,
I will, but it might be a little while before I get to that. I've started my own SDR transceiver project, stealing much from cutesdr, but with code that I find more readable. The world doesn't need this, but I need to learn. I'm getting it working first with the FunCube dongle, but will later move to Softrocks then Ozy.

Like I said, I'll give your fork a go eventually, and will get back to you with my findings.
Cheers.